RIOT-OS/RIOT #6642: xtimer: broken on 'native` when with thread_flags

haukepetersen, 2017-02-23:
refining #6624

The following code hangs after the first timeout on native:
```c
static void time_evt(void *arg)
{
    thread_flags_set((thread_t *)arg, 0x1);
}

int main(void)
{
    xtimer_t timer;
    timer.callback = time_evt;
    timer.arg = (void *)sched_active_thread;

    while (1) {
        xtimer_set(&timer, 1000000);
        thread_flags_wait_any(0x1);
        printf("+++ timeout +++\n");
    }

    return 0;
}
```

After a quick debugging session I might have found a lead: it seems like native is doing an early context switch from its interrupt context back to the main thread when calling `thread_flags_set()`. The second call to `xtimer_set()` exits in `_lltimer_set()` without setting the new timer, because `_in_handler` is still set. If I see it correctly, the `_in_handler` being set because the `_timer_callback()` function never actually exists -> hence it seems like the switch from interrupt to thread context is happening early.

I am not sure however, how this should be fixed?! Are there maybe some `if (in_isr())` checks missing in the thread_flags implementation?

haukepetersen, 2017-02-23:
The problem seems to be broken scheduling behavior in native, `isr_thread_yield()` (in `native_cpu.c`) calls the scheduler directly, and does not allow the interrupt to run to completion. If I am wrong or is that quite broken behavior?!

gebart, 2017-02-24:
@haukepetersen I agree that the ISRs must be allowed to run to finish before switching contexts.

lebrush, 2017-02-24:
I faced a similar situation when implementing the `xtimer_mutex_lock_timeout` and solved by adding a [`thread_yield_higher()`](https://github.com/RIOT-OS/RIOT/blob/master/sys/xtimer/xtimer.c#L242) (which then I figured out it should rather be a [`sched_context_switch_request = 1`](https://github.com/RIOT-OS/RIOT/pull/6441/files#diff-7412db2749b270423cfd5430e69744b9R253) since it's called from an interrupt context) in the callback. 

Maybe that was then a native pitfall after all but IIRC @OlegHahm saw the same behaviour in a SAM platform.

haukepetersen, 2017-02-27:
>  ISRs must be allowed to run to finish before switching contexts.

Agree, this is not only a **shoul**, but a **MUST**! 

So it seems this behavior is (to some extend) related to #6442. The question is how we can fix this once and for all... IMHO this should be a high priority thing to do, but I don't have the know-how to fix this. Anyone else volunteering?

OlegHahm, 2017-02-27:
Maybe I don't it right, but why do you assume that the ISR does not run to completion? I can't neither see it from the native code (just calling `sched_run()`) does not immediately switch context) nor from putting a debug puts after `thread_yield_higher()` in `_thread_flags_wait()`.

haukepetersen, 2017-02-27:
I'll put together a test application to demonstrate this behavior.

haukepetersen, 2017-02-27:
done: #6659

kaspar030, 2017-02-27:
#6660 fixes the issue for me.

PR #6660 (2017-03-02): cpu: native: fix thread_yield_higher isr_is_in() case
Previously, native created a new "setcontext" context whenever ```thread_yield_higher()``` was called. That's unnecessary and broken. At the end of interrupt our interrupt handlers, ```sched_context_switch_request``` is checked and triggers an eventually needed context switch.

Fixes #6642.
closed 2017-03-02 by OlegHahm