RIOT-OS/RIOT #6646: Problems trying to build RIOT on Tinycore

eprive, 2017-02-23:
Encountering errors when trying to build RIOT on Tinycore 7.2 (with gcc package loaded)

- after loading iproute2 (required for the ip command) reports : "RTLINK reports: Operation not supported"
- After make all : "./--version: error while loading shared libraries .....
- After make term : "./--version: error while loading shared libraries .....

But the executable starts: "/home/tc/RIOT-master/examples/default/bin/native/default.elf tap0"
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
etc.
: ioctl TUNSETIFF: Operation not permitted
: probably the tap interface ()  does not exist ....(etc.)

Any ideas ?

kaspar030, 2017-02-24:
> Encountering errors when trying to build RIOT on Tinycore 7.2 (with gcc package loaded)

Interesting! :)

> after loading iproute2 (required for the ip command) reports : "RTLINK reports: Operation not supported"

is the tuntap module loaded (check lsmod output...)?

> After make all : "./--version: error while loading shared libraries .....
> After make term : "./--version: error while loading shared libraries .....

Weird. The only places where native uses "--version" seems to be gcc or ldd.

Could you try

    # gcc --version
    # ldd --version

... and see if they output their version if called manually?

> But the executable starts: "/home/tc/RIOT-master/examples/default/bin/native/default.elf tap0"
> RIOT native interrupts/signals initialized.
> LED_RED_OFF
> LED_GREEN_ON
> etc.
> : ioctl TUNSETIFF: Operation not permitted
> : probably the tap interface () does not exist ....(etc.)

It says "Operation not permitted", maybe try as root? If iproute2 failed to create the tap interface above, RIOT tries to create it itself. That only succeeds if run as root. (Though running as root is not recommended.)

eprive, 2017-02-24:
gcc --version -> 5.2.0
ldd --version -> "./--version error while loading shared libraries ....etc"

It could well be that "ip" doesn't support the bridge functionality in Tinycore. Many commands and applications have been stripped in Tinycore to reduce the footprint.

smlng, 2019-01-22:
closing due to inactivity.

closed 2019-01-22 by smlng