RIOT-OS/RIOT #6648: gnrc: configure global IP addr on startup (for debugging)

haukepetersen, 2017-02-24:
I just had an idea that would make my life (and hopefully that of others) a little easier, but I don't quite know where to implement it:

It would be nice to be able to define a (global) IPv6 address in the Makefile, that is added to the default (e.g. the first) network interface. For example something like:
```
GNRC_DBG_ADD_ADDR = "2001:affe:0815"
```

miri64, 2017-02-27:
See https://github.com/RIOT-OS/RIOT/pull/6623?

smlng, 2017-02-27:
I'm unsure if configuring _global_ IPv6 addresses will work without a 6L(B)R in the network. A while back (ages ago) I flashed 2 boards with RIOT gnrc_networking, manually configured global IPv6 address (or ULAs) but couldn't ping between them, because ND couldn't resolve IP addresses to MAC addresses. Maybe setting ncache entries manually, too, would help?

miri64, 2017-02-27:
> Maybe setting ncache entries manually, too, would help?

You need to set some kind of default route, yes.

smlng, 2017-02-27:
@miri64 I doubt  a default route suffices - I think the problem is remains to resolve L3 to L2 address <- neighbor discovery. IIRC having 2 RIOT nodes with manually setup global IPv6 addresses gets stuck in ND, because there will be no NA to an initial NS by one node for the IP address of the other.

OlegHahm, 2017-03-02:
Well, at least for a non-6LoWPAN scenario this should work anyway.

haukepetersen, 2020-09-07:
Its been a long time and (at least I personally) don't really have the need for the above described use case anymore. Also, for small desk-style setups I guess #6623 should suffice.

So from my perspective I consider this issue solved - feel free to re-open if you feel different.

closed 2020-09-07 by haukepetersen