RIOT-OS/RIOT #6650: GPIO Port shift in cc2538 is unconsistent

AnonMall, 2017-02-24:
Hi,
I just realized when trying to initialize a GPIO Pin with an interrupt through `gpio_init_int(...)` the pin enumaration is inconsistent to the defines in `cc2538_gpio.h`. While the port shift in `cc2538_gpio.h` is 3, it is 12 in `periph/gpio.c`. So when trying to initialize the GPIO `GPIO_PB0` for instance, which is the gpio with port=1 and pin=0 (so encoding by `cc2538_gpio.h`means its pin number 8) `gpio_init_int(...)` will try to access `isr_ctx[][]` at 4294967295, as `port_num(...)` also subracts 1. This leads to a hardfault on my re-mote platform.

smlng, 2017-08-30:
will be resolved with #7517 and finally #7320

aabadie, 2018-11-05:
@smlng, #7517 is merged but #7320 is closed. Can we consider this issue as resolved and close it or should we wait ?

smlng, 2018-11-05:
yeah should be fixed, but the implementation could benefit from some cleanup - I've a WIP branch on that. Anyway, this can be closed.

closed 2018-11-05 by smlng