RIOT-OS/RIOT #6657: gcoaps example in RIOT source code

pithadiyarp, 2017-02-27:
hi ,,
can anyone tell me how to test gcoaps example for samr21 provide on RIOT code .
my query -
1) for this example how many samr21 board is Required (two or one) .

Thanks..

miri64, 2017-02-27:
If you want to communicate with it you need at least two (or one and some kind of border router device like a Raspberry-Pi with a WPAN radio device attached). It would be great if you could give us a pointer to where the [README](https://github.com/RIOT-OS/RIOT/blob/master/examples/gcoap/README.md) of that example is lacking in that regard.

kb2ma, 2017-02-27:
In addition to @miri64's suggestions, see [README-slip.md](https://github.com/RIOT-OS/RIOT/blob/master/examples/gcoap/README-slip.md) specific to the use of a samr21-xpro board in a border router setup, connected via USB/serial to a Linux machine.

pithadiyarp, 2017-02-28:
hi ,
in gcoap example coap get command
ex: coap get ip port path data 
in path what should i provide that i didn't understand (for native platform only).
details -
--> in that i created two taps like tap0 and tap1. now i want to transfer data from tap1 to tap0 and ping is success fully work that i verified with Wireshark tool .

kb2ma, 2017-02-28:
gcoap includes two built-in paths:

1. `/.well-known/core` -- the standard CoAP discovery path
2. `/cli/stats` -- simple statistics about the CoAP CLI itself; reports on count of requests the CLI has sent

Below is a transcript from my workstation. The last command shows the GET request. `ifconfig` shows how to determine the IP address of the process, and `coap info` shows the listening port and the count of requests sent.

```
kbee@gazelle:~/dev/riot/repo/examples/gcoap$ bin/native/gcoap.elf tap0
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2017.01-devel-753-gb7e25-gazelle-gcoap/observe)
gcoap example app
All up, running the shell now
> ifconfig
ifconfig
Iface  6   HWaddr: f6:1c:87:f3:a4:a7 
           
           MTU:1500  HL:64  
           Source address length: 6
           Link type: wired
           inet6 addr: ff02::1/128  scope: local [multicast]
           inet6 addr: fe80::f41c:87ff:fef3:a4a7/64  scope: local
           inet6 addr: ff02::1:fff3:a4a7/128  scope: local [multicast]
           
> coap info
coap info
CoAP server is listening on port 5683
 CLI requests sent: 0
CoAP open requests: 0
> coap get fe80::88e3:51ff:febe:f54d 5683 /.well-known/core
coap get fe80::88e3:51ff:febe:f54d 5683 /.well-known/core
gcoap_cli: sending msg ID 58882, 23 bytes
> gcoap: response Success, code 2.05, 12 bytes
</cli/stats>
```

pithadiyarp, 2017-02-28:
Hi i tried this but is given me some error -

root@CPU-262U:~/RIOT/examples/gcoap# bin/native/gcoap.elf tap0
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2017.04-devel-48-g0b70-CPU-262U)
gcoap example app
All up, running the shell now
> ifconfig
ifconfig
Iface  6   HWaddr: be:1e:3c:7d:c7:b1 
           
           MTU:1500  HL:64  
           Source address length: 6
           Link type: wired
           inet6 addr: ff02::1/128  scope: local [multicast]
           inet6 addr: fe80::bc1e:3cff:fe7d:c7b1/64  scope: local
           inet6 addr: ff02::1:ff7d:c7b1/128  scope: local [multicast]
           
> coap info
coap info
CoAP server is listening on port 5683
 CLI requests sent: 0
CoAP open requests: 0
> coap get fe80::bc1e:3cff:fe7d:c7b1 5683 /.wellknown/core
coap get fe80::bc1e:3cff:fe7d:c7b1 5683 /.wellknown/core
gcoap_cli: sending msg ID 58882, 22 bytes
gcoap: response Error, code 4.04, empty payload

can you please give me solution for the same ...

kb2ma, 2017-02-28:
There is a '-' in `/.well-known/core`. ;-)

pithadiyarp, 2017-02-28:
hi ,,
I created two tap tap0 and tap1
can you please tell me how can i transfer data between two tap . tap detailsgiven below...

# TAP1
 root@CPU-262U:~/RIOT/examples/gcoap# bin/native/gcoap.elf tap1
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2017.04-devel-48-g0b70-CPU-262U)
gcoap example app
All up, running the shell now
> ifconfig
ifconfig
Iface  6   HWaddr: b6:65:f3:91:22:fd 
           
           MTU:1500  HL:64  
           Source address length: 6
           Link type: wired
           inet6 addr: ff02::1/128  scope: local [multicast]
           inet6 addr: fe80::b465:f3ff:fe91:22fd/64  scope: local
           inet6 addr: ff02::1:ff91:22fd/128  scope: local [multicast]
           
> 

# TAP0
root@CPU-262U:~/RIOT/examples/gcoap# bin/native/gcoap.elf tap0
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2017.04-devel-48-g0b70-CPU-262U)
gcoap example app
All up, running the shell now
> ifconfig
ifconfig
Iface  6   HWaddr: be:1e:3c:7d:c7:b1 
           
           MTU:1500  HL:64  
           Source address length: 6
           Link type: wired
           inet6 addr: ff02::1/128  scope: local [multicast]
           inet6 addr: fe80::bc1e:3cff:fe7d:c7b1/64  scope: local
           inet6 addr: ff02::1:ff7d:c7b1/128  scope: local [multicast]
           
> coap info
coap info
CoAP server is listening on port 5683
 CLI requests sent: 0
CoAP open requests: 0
> coap get fe80::bc1e:3cff:fe7d:c7b1 5683 /.well-known/core
coap get fe80::bc1e:3cff:fe7d:c7b1 5683 /.well-known/core
gcoap_cli: sending msg ID 58882, 23 bytes
gcoap: response Success, code 2.05, 12 bytes
</cli/stats>
> coap info
coap info
CoAP server is listening on port 5683
 CLI requests sent: 1
CoAP open requests: 0
> coap get fe80::b465:f3ff:fe91:22fd 5683 /.well-known/core
coap get fe80::b465:f3ff:fe91:22fd 5683 /.well-known/core
gcoap_cli: sending msg ID 59138, 23 bytes
gcoap: timeout for msg ID 59138
>

pithadiyarp, 2017-03-01:
hi ,,
Thanks it's very helpful to me...

kb2ma, 2017-03-01:
Excellent! Feel free to close this issue then. You always can open another one or send a message on the riot-users mailing list for more questions.

closed 2017-03-01 by pithadiyarp