RIOT-OS/RIOT #6669: boards, nucleo: THREAD_STACKSIZE_DEFAULT too large

smlng, 2017-03-01:
In #6662 I optimized the size of `tests/xtimer_drift` to fit onto board _arduino duemilanove_, which has only 2K RAM. In the Makefile of this test the `nucleo32-f031` and `nucleo32-f042` are blacklisted due to insufficient mem - and @miri64 asked it this is still valid for that tests, so I looked into it.

Simply running make for these boards, resulted in a linker error saying that section .bss would not fit into RAM, with an overflow of 472B for f042 and 2512B for f031 respectively. Btw. f042 has 6k RAM and f032 4K, so 2-3 times more than the arduino. 

So why does the test not fit onto the boards, as they sure have enough RAM? Well, the problem is all nucleos share the definition of `THREAD_STACKSIZE_DEFAULT` in `cpu/cortexm_common` where it is set to 1024B (+ extra 512B for printf) <- so not surprisingly 4K RAM (f031) are not much for 3 threads and some data.

The two nucleos are blacklisted for many tests and examples, which might just due to the fact of arbitrary large stack sizes. Question to those more experiences with these boards, is there any reason the stack size has to be that large? IMHO it should be adapted for such constrained devices.

See here:
```
./driver_pir/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./gnrc_sock_ip/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./gnrc_sock_udp/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f042 nucleo32-f031
./irq/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./libfixmath_unittests/Makefile
BOARD_INSUFFICIENT_MEMORY := weio nucleo32-f042 nucleo32-f031
./lwip_sock_tcp/Makefile
BOARD_INSUFFICIENT_MEMORY = nucleo-f030 nucleo32-f031 nucleo32-f042 nucleo-f334 \
./lwip_sock_udp/Makefile
BOARD_INSUFFICIENT_MEMORY = nucleo32-f042 nucleo-f030 nucleo32-f031 nucleo-f042 \
./msg_send_receive/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./msg_try_receive/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./mutex_order/Makefile
BOARD_INSUFFICIENT_MEMORY := stm32f0discovery weio nucleo-f030 nucleo32-f042 nucleo32-f031
./mutex_unlock_and_sleep/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./netdev2_test/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./periph_uart/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./pipe/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./pkg_u8g2/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./pthread_condition_variable/Makefile
BOARD_INSUFFICIENT_MEMORY := stm32f0discovery nucleo32-f031
./pthread_cooperation/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./rmutex/Makefile
BOARD_INSUFFICIENT_MEMORY := stm32f0discovery weio nucleo-f030 nucleo32-f031 \
./sched_testing/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./thread_basic/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./thread_exit/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./thread_flags/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./thread_msg/Makefile
BOARD_INSUFFICIENT_MEMORY := stm32f0discovery nucleo32-f042 nucleo32-f031
./thread_msg_block_w_queue/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./thread_msg_block_wo_queue/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./thread_msg_seq/Makefile
BOARD_INSUFFICIENT_MEMORY := stm32f0discovery nucleo32-f042 nucleo32-f031
./xtimer_drift/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f042 nucleo32-f031
./xtimer_hang/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
./xtimer_longterm/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f042 nucleo32-f031
./xtimer_msg/Makefile
BOARD_INSUFFICIENT_MEMORY := nucleo32-f031
```

aabadie, 2017-03-01:
I remember that we had this discussion when @vincent-d ported the nucleo32-f042 in #6273. See comments [here](https://github.com/RIOT-OS/RIOT/pull/6273#issuecomment-270658233), [here](https://github.com/RIOT-OS/RIOT/pull/6273#issuecomment-270907622) and the [winning comment](https://github.com/RIOT-OS/RIOT/pull/6273#issuecomment-270909774) of @OlegHahm.
Maybe there's still room for debate.

smlng, 2017-03-01:
mhm, I'm not familiar with the differences within the cortex (m0, m1, m2, ...) family. So I don't know if some may require larger stack sizes because they bring more stuff. However, I would guess it makes sense to adapt stack sizes by minor family, i.e. m0, m1 ... . At least it seams that boards with m0 very likely have limited RAM, too.

Or we could define `THREAD_STACKSIZE_DEFAULT` within Makefile of tests, where we know how much ram this app/test needs?

kaspar030, 2017-03-01:
> is there any reason the stack size has to be that large? IMHO it should be adapted for such constrained devices.

The stack sizes needed are architecture (and application-)dependent. That's why AVR8 and MSP430 have smaller "default" stacksizes than ARM, and all ARM (which all compile to thumb) have the same, as all applications compiled for any ARM board will need the same amount of stack.

Setting smaller default stacksizes for ARM boards with little memory will very probably cause problems.

If a test needs less stacksize, it should not use the default defines.

@miri64 didn't you propose defines for small, medium, large stacksizes some time ago?

smlng, 2017-03-01:
@kaspar030 so you would favor what I said last:

> Or we could define THREAD_STACKSIZE_DEFAULT within Makefile of tests, where we know how much ram this app/test needs?

which is probably the best thing to do, as its mostly app/test depended how much mem is actually required, having some predefined values (MIN, MED, LRG, XXL) might be a good starting point.

kaspar030, 2017-03-01:
> define THREAD_STACKSIZE_DEFAULT within Makefile of tests

Not the best idea, as many modules use that define for their stacks. Tweaking it for the main application could break those.

IMHO setting ```THREAD_STACKSIZE_MAIN``` is the way to go for the main thread, as there's no way to change the define in code.

For threads spawned from within the application, MIN SML DFLT LARGE should be used.

Unfortunately, some modules also use THREAD_STACKSIZE_MAIN, those would have to be changed to DEFAULT.

smlng, 2017-03-01:
> > define THREAD_STACKSIZE_DEFAULT within Makefile of tests
>
> Not the best idea, as many modules use that define for their stacks. Tweaking it for the main application could break those.

@kaspar030 good point. 

Okay lets go by example, see `tests/xtimer_hang` for instance: where we have those 2 very tiny background timer threads, which do not require a large stack. 

Using THREAD_STACKSIZE_DEFAULT or MAIN here, would waste memory. And even redefining THREAD_STACKSIZE_MAIN is bounded by the size the actual main requires and will still be larger than necessary for these background tasks. Still it would be good to have some reasonable default sizes at hand, which might be relative to THREAD_STACKSIZE_DEFAULT and can be overwritten on demand?

kaspar030, 2017-03-01:
> Using THREAD_STACKSIZE_DEFAULT or MAIN here, would waste memory.

Would DEFAULT/2 work on all platforms? That's how we'd define e.g., ```THREAD_STACKSIZE_SMALL```.

aabadie, 2017-03-15:
Any update on this one ? I started to work on a port of RIOT on nucleo-l011k4 which has only 2K of RAM. With the actual default stack size, even hello-world doesn't build...

kaspar030, 2017-03-15:
Maybe it's time for a blinker app ;)

Seriously, 2k ram is *very* little, especially for ARMs. Try to get the minimal example working first, maybe add some output using fmt...

smlng, 2017-03-15:
2K RAM is enough for most tests (if they are properly implemented, without unnecessary bloat). At least those _without_ networking should work; that's my experience with an _arduino duemilanove_ with 2K RAM.

smlng, 2017-03-15:
@aabadie you could specify distinct stack sizes for this board, overwriting those of cortex-common ...

kaspar030, 2017-03-15:
> that's my experience with an arduino duemilanove with 2K RAM.

The situation is quite different on 8bit platforms...

kaspar030, 2017-03-15:
> @aabadie you could specify distinct stack sizes for this board, overwriting those of cortex-common ...

Just as a reminder: all ARM boards need the same amount of stack space. So if lowering the default stack size works for nucleo-l011k4, it works for all other ARM boards, too (not considering stack usage differences of e.g., periph).

smlng, 2017-03-15:
>The situation is quite different on 8bit platforms...

that might be true, but this primarily  about low-mem/constraint nucleo(32) boards , which are (mostly) 32 Bit, right?

> all ARM boards need the same amount of stack space.

good to know; thought there might be (some) differences between cortex m0, ..., m3, m4

closed 2019-09-10 by stale[bot]