RIOT-OS/RIOT #6680: tests: unclear semantic for thread_msg_block_wo_queue

OlegHahm, 2017-03-02:
I don't understand the basic idea of the test application `thread_msg_block_wo_queue`. Unfortunately, it misses a README or an expect script and the output is also not very enlightening.

The application spawns a second thread, waits for a message, that is then sent by the second thread. This thread (with a higher priority) then tries to send another message and waits for a response. After receiving the first message in the first thread, this thread terminates letting the second state in REPLY_BLOCKED forever. Is this the intension? What should be tested? And why?

OlegHahm, 2017-03-02:
I just realized that `tests_msg_block_w_queue` is similar and as unclear for me.

Lotterleben, 2017-03-02:
The corresponding issue to these tests was https://github.com/RIOT-OS/RIOT/issues/100. If I remember it correctly, the point of the test was to ensure correct blocking behaviour while a thread is busy sending a message… Is this a call for us to write a README? ;) I'd be happy to have a go at that on friday.

OlegHahm, 2017-03-02:
Well, in the end it would be terrific to have a README, a pexpect script for automation, and inclusion in the test framework (i.e., enabling the `make test` target). But for starters a README would be nice, yes. :)

Lotterleben, 2017-03-02:
OK, I'll have a go at it after work. :)