RIOT-OS/RIOT #6686: SAML21 SPI broken

dylad, 2017-03-02:
Hello,
I was doing some tests for #6645 with my saml21-xplained board but I'm facing a SPI issue.
It appears SPI for SAML21 is broken. I cannot leave spi_acquire function.
I also tried with tests/periph_spi, shell is blocked in a loop and SPI is not initialized.
I will try to investigate further and come up with a PR to fix it.

dylad, 2017-03-02:
I figured out the problem.
SERCOM ID calculation located in cpu/sam0_common/include/periph_cpu_common.h is wrong for SAML family. 
SERCOM0 is located at  0x42000800 for SAMD family and SERCOM0 is at 0x42000000UL for SAML family. 
As this calculation is address related. It is obviously wrong for SAML.
#6687 fixes the problem.

aabadie, 2017-03-21:
As you referenced this issue in #6437. I think we can safely close this one.

closed 2017-03-21 by aabadie