RIOT-OS/RIOT #6715: Support For cc1310 and cc1350

denzelwamburu, 2017-03-07:
Am trying to port this to TI's cc1310 and cc1350 family. 

Is there to do this?
The hello world doesn't seem to work on this.

Thanks.

PeterKietzmann, 2017-03-28:
Hi @denzelwamburu and welcome to RIOT! Can you give some more details about what has been done? Maybe share a branch or open PR with your changes?

cvgaviao, 2017-10-19:
Hi, I'm also interested to use RIOT on CC1310/CC1350 Launchpads and my custom boards with those ICs. 
Anyone have tried to use RIOT/boards/cc2650-launchpad on one of them ?

denzelwamburu, 2017-10-20:
@cvgaviao  I tried but to no avail. My cc1310 board seems it didn't have enough memory

cvgaviao, 2017-10-20:
@denzelwamburu, I will have a try this weekend. 
But enought memory makes no sense because I'm running Contiki on my LaunchPads and what I have read is that Riot is smaller than contiki.

PeterKietzmann, 2017-11-27:
These wiki pages might help you:
- [Porting Guide](https://github.com/RIOT-OS/RIOT/wiki/Porting-Guide)
- [Writing a device driver inRIOT](https://github.com/RIOT-OS/RIOT/wiki/Guide:-Writing-a-device-driver-in-RIOT)
@denzelwamburu, @cvgaviao can you give a quick status update about your efforts?

lafka, 2018-02-18:
I can confirm that CC2650 example works fine for me. Only step i've done is to copy related files from cc2650 in cpu and board directories to their own cc13xx dirs.

Marc-Aurele, 2018-03-02:
Hi,
I'm also very interested in using RIOT on my CC1310 launchpad. I'm just discovering this MCU and it seems to be full of promise. Usually, im using RIOT on my STM32.

PeterKietzmann, 2018-08-22:
@Marc-Aurele, @lafka, @cvgaviao, @wamburu any progress since February?

PeterKietzmann, 2019-02-21:
@maribu no that you seem to be the CC1xxxx radio expert I just wanted you to be aware of this feature request.

PeterKietzmann, 2019-07-17:
It seems there is no ongoing work nor interest in this so I'm closing the issue as it seems very outdated.

closed 2019-07-17 by PeterKietzmann