RIOT-OS/RIOT #6721: nucleo-f401 hardware rev. c-01: no high speed clock source

jbeyerstedt, 2017-03-08:
Some board revisions of the nucleo-f401re fail at initializing the cpu clock in the `cpu_clock_init` function of cpu.c (https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32f4/cpu.c).

The board MB1136 C-03 works well, but my C-01 board does not get farther than this line
```c
/* wait for the high speed clock source to be ready */
while (!(RCC->CR & CLOCK_CR_SOURCE_RDY)) {}
```
because, the X2 crystal is not present on this particular board revision.

Is there an easy way to detect the board revisions and change the clock source accordingly?
Otherwise it would be nice, if there is a hint in the wiki (that only revision C-03 works).

lebrush, 2017-03-13:
I think that through some solder bridges you can use an exported clock from the st-link and feed it as a HSE into the stm32 (at least this is possible with the nucleo f103).

See http://www.st.com/resource/en/user_manual/dm00105823.pdf section 6.7.1 for more details.

Of course, you can also use the internal HSI as a clock source. In the [`stm32f1`](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32f1/cpu.c) mcu depending whether `CLOCK_HSE` or `CLOK_HSI` are defined in `board.h` a different clock source is selected. 

I hope it helps.

jbeyerstedt, 2017-03-15:
Thanks for this information, that I somehow skipped during reading the Nucleo-64 user manual. But the greater problem is actually the 32 KHz clock supply, which is also not present in this particular Board revision. So the RTC peripheral cannot be used (at cpu/stm32_common/periph/rtc.c):
```c
/* Enable the LSE clock (external 32.768 kHz oscillator) */
RCC->BDCR &= ~(RCC_BDCR_LSEON);
RCC->BDCR &= ~(RCC_BDCR_LSEBYP);
RCC->BDCR |= RCC_BDCR_LSEON;
while ((RCC->BDCR & RCC_BDCR_LSERDY) == 0) {}
```
The `RCC_BDCR_LSERDY` flag never gets set.

But I managed to get the default example running by removing the RTC peripheral from the `Makefile.features` of the nucleo-f401 board. 

Should I add a comment to the wiki, that only revision C-02 and later will work with the default configuration?

lebrush, 2017-03-15:
That would be great! Thank you

closed 2017-03-16 by jbeyerstedt