RIOT-OS/RIOT #6724: Couldn't make hello world work on cc2650stk

chenek, 2017-03-09:
I can run hello world example successfully on my CC2538DK. However, when I build hello world example with "make BOARD=cc2650stk" and download elf file to my CC2650STK. I cannot see any output from UART. Do I miss anything?

smlng, 2017-03-10:
@chenek thanks for reporting, I'll try to check in the next days.

chenek, 2017-03-10:
Look forward to hearing feedback from you.

shaolinkhoa, 2017-03-14:
Hi, I predefined cc2650stk so that RIOT OS can run on cc2650 launchpad :
**/home/ntuankhoa/RIOT/boards/cc2650stk/include/periph_conf.h**
```
#define UART_RX_PIN         (2)  
#define UART_TX_PIN         (3)
```

**/home/ntuankhoa/RIOT/boards/cc2650stk/include/board.h**
```
#define LED0_PIN            GPIO_PIN(0, 6) //red
#define LED1_PIN            GPIO_PIN(0, 7) //green
```

I cloned default example to my_project and added:
```
#include "board.h"
int led_0_on(int argc, char **argv) { //red
    (void)argc;
    (void)argv;
    LED0_ON;
    printf("LED_RED_ON\n");
    return 0;
}
int led_0_off(int argc, char **argv) { //red
    (void)argc;
    (void)argv;
    LED0_OFF;
    printf("LED_RED_OFF\n");
    return 0;
}
const shell_command_t shell_commands[] = {
    {"L0_on", "turn led 0 on", led_0_on},
    {"L0_off", "turn led 0 off", led_0_off},
    { NULL, NULL, NULL }
};
int main(void)
{
    printf("test uart 1\n");
#ifdef FEATURE_PERIPH_RTC
    rtc_init();
#endif
#ifdef MODULE_NETIF
    gnrc_netreg_entry_t dump = GNRC_NETREG_ENTRY_INIT_PID(GNRC_NETREG_DEMUX_CTX_ALL,
                                                          gnrc_pktdump_pid);
    gnrc_netreg_register(GNRC_NETTYPE_UNDEF, &dump);
#endif
    LED_RED_ON;
    LED_GREEN_ON;
    (void) puts("Welcome to RIOT!");
    printf("test uart 2\n");
    char line_buf[SHELL_DEFAULT_BUFSIZE];
    shell_run(shell_commands, line_buf, SHELL_DEFAULT_BUFSIZE);
    return 0;
}
```

After I loaded my_project into cc2650 launchpad:
+ 2 led Red & Green turn on
+ Receive from terminal : 
```
main(): This is RIOT! (Version: 2017.04-devel-360-g408d1-ntuankhoa)
test uart 1
Welcome to RIOT!
test uart 2
```
but I can't send my shell command to cc2650 launch pad. I sent "help" but nothing happened.
Native mode is working fine. I can send shell commands and control leds.    
Did I miss anything ?

kaspar030, 2017-03-14:
@chenek Which serial port is the cc2650 connected to? (unplug/replug and see if ```dmesg``` gives a hint)

Maybe you need to specify the right serial port by adding "PORT=/dev/<port>" to your command line.

chenek, 2017-03-15:
I connect my CC2650STK on Win & and use teraterm with baudrate 115200. I cam pretty sure the connection is good. If I download Contiki OS into my CC21650STK, it can output to UART without problem.

shaolinkhoa, 2017-03-15:
Hi,
Are there any examples about shell commands in UART ? or the shell commands only works on native mode ?

nRhR, 2017-05-31:
@chenek Hey there, I have the same problem with the cc2650stk RIOT shell response.
 Is there any progress?

chenek, 2017-05-31:
No, I have no progress on this.

A-Paul, 2017-06-02:
Hi @chenek. I can't reproduce your problems here with my sensortag (under Debian).
> 2017-06-02 19:08:29,269 - INFO # Connect to serial port /dev/ttyACM0
Welcome to pyterm!
Type '/exit' to exit.
2017-06-02 19:08:38,367 - INFO # main(): This is RIOT! (Version: 2017.04-devel-1135-g3b725-mobi31)
2017-06-02 19:08:38,368 - INFO # Hello World!
2017-06-02 19:08:38,372 - INFO # You are running RIOT on a(n) cc2650stk board.
2017-06-02 19:08:38,375 - INFO # This board features a(n) cc26x0 MCU.

What do you use to flash, TI's uniflash?
Do you have the terminal already started when flashing?

chenek, 2017-06-03:
@A-Paul I just git to the latest source code and do "make BOARD=cc2650stk" on my VirtualBox Ubuntu 14.04. I still cannot see UART output after I download hex or elf to my CC2650STK. I attach my hex/elf for your reference. Maybe you can attach yours so I can download it to my CC2650STK to see if it works on my HW.

[cc2650stk.zip](https://github.com/RIOT-OS/RIOT/files/1049443/cc2650stk.zip)

nRhR, 2017-06-03:
Same here. 
maybe i missed some elementary, but if i run my code on native i get response.
lets keep on trying :)

nRhR, 2017-06-03:
My setup: @chenek @A-Paul 

-code 
-output native
-output BOARD=cc2650

`/*
 * Copyright (C) 2008, 2009, 2010  Kaspar Schleiser <kaspar@schleiser.de>
 * Copyright (C) 2013 INRIA
 * Copyright (C) 2013 Ludwig Knüpfer <ludwig.knuepfer@fu-berlin.de>
 *
 * This file is subject to the terms and conditions of the GNU Lesser
 * General Public License v2.1. See the file LICENSE in the top level
 * directory for more details.
 */

/**
 * @ingroup     examples
 * @{
 *
 * @file
 * @brief       Default application that shows a lot of functionality of RIOT
 *
 * @author      Kaspar Schleiser <kaspar@schleiser.de>
 * @author      Oliver Hahm <oliver.hahm@inria.fr>
 * @author      Ludwig Knüpfer <ludwig.knuepfer@fu-berlin.de>
 *
 * @}
 */

#include <stdio.h>
#include <string.h>

#include "thread.h"
#include "shell.h"
#include "shell_commands.h"
#include "board.h"

#if FEATURE_PERIPH_RTC
#include "periph/rtc.h"
#endif

#ifdef MODULE_NETIF
#include "net/gnrc/pktdump.h"
#include "net/gnrc.h"
#endif

int hello_world(int argc, char **argv) {
    /* Suppress compiler errors */
    (void)argc;  // number of parameters. 
    (void)argv;  // value of each parameters . Ex: argv[0] argv[1]
    // printf("hello world!\n");
    puts("hello world!\n");
    return 0;
}

const shell_command_t shell_commands[] = {
    {"hello", "prints hello world", hello_world},
    { NULL, NULL, NULL }
};

int main(void)
{
    printf("test uart 1\n");

#ifdef FEATURE_PERIPH_RTC
    rtc_init();
#endif

#ifdef MODULE_NETIF
    gnrc_netreg_entry_t dump = GNRC_NETREG_ENTRY_INIT_PID(GNRC_NETREG_DEMUX_CTX_ALL,
                                                          gnrc_pktdump_pid);
    gnrc_netreg_register(GNRC_NETTYPE_UNDEF, &dump);
#endif

    (void) puts("Welcome to RIOT!");
     printf("test uart 2\n");

    char line_buf[SHELL_DEFAULT_BUFSIZE];
   // shell_run(NULL, line_buf, SHELL_DEFAULT_BUFSIZE);
    shell_run(shell_commands, line_buf, SHELL_DEFAULT_BUFSIZE);

    return 0;
}`

native:
[image]

BOARD=cc2650:
[image]

nRhR, 2017-06-03:
the code upload is messed up.
i will try it better if needed

A-Paul, 2017-06-07:
Hi @chenek, I tried to use your image. Here is the output:

> 2017-06-07 02:42:30,735 - INFO # main(): This is RIOT! (Version: 2017.04-devel-1200-g3cdb-yk-VirtualBox)
2017-06-07 02:42:30,736 - INFO # Hello World!
2017-06-07 02:42:30,741 - INFO # You are running RIOT on a(n) cc2650stk board.
2017-06-07 02:42:30,743 - INFO # This board features a(n) cc26x0 MCU.

It seems that your toolchain is working in respect of building an image file. Also there it is no issue in the RIOT codebase.
I'd like to ask you to close this issue and request support on the [RIOT-users mailing list](http://lists.riot-os.org/mailman/listinfo/users).
In your opening post you might provide some information about your image burning and terminal handling workflow.

chenek, 2017-06-07:
That is strange because my CC2650STK has no problem to output UART messages when I test Contiki OS. I would check my CC2650STK and Thanks for verifying my binary.

closed 2017-06-07 by chenek

A-Paul, 2017-06-07:
Hi @chenek. Hope you still get notifications...
Are you aware of the fact that *hello_world* just prints after power on *once* and that there is no reset button on the sensortag defined?
Maybe see you on the ML.

A-Paul, 2017-06-14:
Hi @shaolinkhoa and @nRhR your reports ( https://github.com/RIOT-OS/RIOT/issues/6724#issuecomment-286322009 and https://github.com/RIOT-OS/RIOT/issues/6724#issuecomment-305968053) are in the wrong place, because this issue is about a -non existing- failure *writing* to UART.
Your problems are based on a shell freeze when *reading* from UART. I opened an issue https://github.com/RIOT-OS/RIOT/issues/7188 to resolve this bug. Feel free to help debugging.

@shaolinkhoa, your notice in https://github.com/RIOT-OS/RIOT/issues/6724#issuecomment-286322009 regarding *shell_init()* is valid. I've modified the wiki. Thanks for reporting.