RIOT-OS/RIOT #6750: at86rf233 (openlabs module) driver does not work (on nucleo-f401)

jbeyerstedt, 2017-03-16:
The openlabs 802.15.4 radio will not initialize with my nucleo-f401 and f411 boards.
Because the F411 board only was recently added to RIOT, a wasn't able to test this board with an older release of RIOT, so I switched to the F401.

The program does not get farther than the while statement in the `_set_state` function of `st86rf2xx_getset.c`:
```c
if (state != AT86RF2XX_STATE_RX_AACK_ON) {
    while (at86rf2xx_get_status(dev) != state);
}
```

So I connected a scope to the SPI bus to check the signal levels and there comes the second problem:
In the current master branch, there is no signal on the SCK line.
For the next step, I switched back to the 2016.10 release of RIOT, because I heard, that there were some big changes in the SPI library recently. With this RIOT version, there is a clock signal, but still no proper initialization of the radio. The code was still stuck at the line mentioned above.

If it helps, I attached a photo of the SPI signal with working clock (2016.10 branch):
[image]
This gets send over and over again.

jbeyerstedt, 2017-03-17:
Sorry, I just realized, that the issue with the missing clock signal is known in issue #6501. But the rest of this issue remains a problem.

thomaseichinger, 2017-03-17:
@jbeyerstedt Thanks for reporting this! Could you provide us with some more information, especially which state doesn't get reached and what's the state reported by the device?

Something like this should do:

```diff
if (state != AT86RF2XX_STATE_RX_AACK_ON) {
-    while (at86rf2xx_get_status(dev) != state);
+    uint8_t dev_state;
+    while ((dev_state = at86rf2xx_get_status(dev)) != state) {
+        printf("%02x != %02x\n", dev_state, state);
+    }
}
```

jbeyerstedt, 2017-03-20:
The issue seems to be stuck while getting to state 19 (AT86RF2XX_STATE_TX_ARET_ON).
The radio always "responds" with state 0 (`00 != 19`).

thomaseichinger, 2017-03-20:
@jbeyerstedt Thanks for providing more details on this. There seems to be a similar issue in #6730. It seems the openlabs module does something different, I will look into it. In the meanwhile, could you see if the approach described in https://github.com/RIOT-OS/RIOT/pull/6730#issuecomment-287719691 brings any improvements?

jbeyerstedt, 2017-03-21:
Sorry, but it's still the same issue, if I revert #5244.

thomaseichinger, 2017-03-21:
@jbeyerstedt To reproduce the problem properly, could you please share your branch including your configuration with me. Also I'd appreciate the description or picture of how you connected the radio module to the board.

PeterKietzmann, 2017-03-30:
>I just tried it with the suggested patch of #6764, but that does not fix this issue.

PeterKietzmann, 2017-03-31:
@jbeyerstedt I'm on an stm32f4discovery board, the openlabs transceiver connected according to the following pin configuration:
```
#ifndef AT86RF2XX_PARAM_SPI
#define AT86RF2XX_PARAM_SPI         (SPI_DEV(0))
#endif
#ifndef AT86RF2XX_PARAM_SPI_CLK
#define AT86RF2XX_PARAM_SPI_CLK     (SPI_CLK_5MHZ)
#endif
#ifndef AT86RF2XX_PARAM_CS
#define AT86RF2XX_PARAM_CS          (GPIO_PIN(4, 15))
#endif
#ifndef AT86RF2XX_PARAM_INT
#define AT86RF2XX_PARAM_INT         (GPIO_PIN(4, 12))
#endif
#ifndef AT86RF2XX_PARAM_SLEEP
#define AT86RF2XX_PARAM_SLEEP       (GPIO_PIN(4, 14))
#endif
#ifndef AT86RF2XX_PARAM_RESET
#define AT86RF2XX_PARAM_RESET       (GPIO_PIN(4, 13))
#endif
```
Then I build (and flash) the `gnrc_networking` example with: `USEMODULE=at86rf233 BOARD=stm32f4discovery make clean all flash term` and everything works as expected. Can you check on your discovery and nucleo boards if you did something different and otherwise lend me your hardware for a moment?

PeterKietzmann, 2017-04-03:
Works like charm with this configuration:
```
#ifndef AT86RF2XX_PARAM_SPI
#define AT86RF2XX_PARAM_SPI         (SPI_DEV(0))
#endif
#ifndef AT86RF2XX_PARAM_SPI_CLK
#define AT86RF2XX_PARAM_SPI_CLK     (SPI_CLK_5MHZ)
#endif
#ifndef AT86RF2XX_PARAM_CS
#define AT86RF2XX_PARAM_CS          (GPIO_PIN(1,  1))
#endif
#ifndef AT86RF2XX_PARAM_INT
#define AT86RF2XX_PARAM_INT         (GPIO_PIN(1, 13))
#endif
#ifndef AT86RF2XX_PARAM_SLEEP
#define AT86RF2XX_PARAM_SLEEP       (GPIO_PIN(1, 15))
#endif
#ifndef AT86RF2XX_PARAM_RESET
#define AT86RF2XX_PARAM_RESET       (GPIO_PIN(1, 14))
#endif
```
I tried with @jbeyerstedt's hardware. Most likely it was just an incorrect wire connection.

closed 2017-04-03 by PeterKietzmann

jbeyerstedt, 2017-04-03:
Just a a follow up:
I tried the configuration of @PeterKietzmann with the state of the master branch I used when opening this issue and it does not work. But something in the last commits to the master branch has somehow fixed this problem. So it was not just a wiring issue.
Anyway: Thanks for the support.

PR #6764 (2017-03-30): stm32_common/spi: fix receive after transmit-only error
This line fixes SPI transmissions on at least the faster class of stm32 devices.

The main problem is that there is a slight delay between the SPI transfer register being written and the BUSY bit being set. From the [datasheet of the stm32f405..439](http://www.st.com/resource/en/reference_manual/dm00031020.pdf) (20MB warning)
```
During discontinuous communications, there is a 2 APB clock period delay between the
write operation to SPI_DR and the BSY bit setting. As a consequence, it is mandatory to
wait first until TXE=1 and then until B [... cut]