RIOT-OS/RIOT #6756: pkg: lwip: superfluous/wrong call of sched_switch()

OlegHahm, 2017-03-17:
In `pkg/lwip/contrib/sys_arch.c` the function `sched_switch()` gets called for unclear reasons after creating the main lwip thread. In my understanding this call doesn't have any effect, since a) the lwip thread has the highest priority anyway and b) `sched_switch()` seems rather like a kernel internal function to me.

miri64, 2017-03-17:
Disclaimer: I don't exactly know anymore, why I put it there. The [porting guide doesn't give a hint either](http://lwip.wikia.com/wiki/Porting_for_an_OS#Threads). Maybe I wanted to enforce a switch to the thread just created, but I don't know why (I really should get used to comment everything o_O). However I have some comments on your statements:

> a) the lwip thread has the highest priority anyway and

But that function isn't the "create lwip thread" function, but the *generic* "create new thread" wrapper for lwIP around the system layer. Just because there is a high priority thread created *now* by lwIP doesn't mean that in some future version there will also be a low priority thread in addition to the tcpip thread.

> b) sched_switch() seems rather like a kernel internal function to me.

It isn't documented as such.

OlegHahm, 2017-03-17:
>  Just because there is a high priority thread created now by lwIP doesn't mean that in some future version there will also be a low priority thread in addition to the tcpip thread.

True, but if it gets a lower priority, `sched_switch()` won't schedule it anyway.

> It isn't documented as such.

Also true and the documentation is IMO a bit misleading in fact, but it seems that you rather should have called `thread_yield_higher()` instead. But I don't see the reason for an additional scheduler call here at all.

miri64, 2019-09-24:
Finally fixed in #12296

PR #12296 (2019-09-24): lwip: sys_arch: replace sched_switch() with thread_yield_higher()

### Contribution description
`sched_switch()` is supposed to be used in core only.

This fixes #6756.

### Testing procedure
Running one of the `lwip` tests on `native` should suffice, as they use the function changed. I used `tests/lwip_sock_udp` and `tests/lwip`.

### Issues/PRs references
Fixes #6756
commit 4c88cb5f (2019-09-24): lwip: sys_arch: replace sched_switch() with thread_yield_higher()

`sched_switch()` is supposed to be used in core only.

This fixes #6756.
closed 2019-09-24 by miri64