RIOT-OS/RIOT #6798: Vagrant: latest vbox (5.1.18) and latest vagrant (1.9.3) don't work together

emmanuelsearch, 2017-03-27:
I upgraded, and since then, when I `vagrant up` I get the error:

```
There was an error while executing `VBoxManage`, a CLI used by Vagrant
for controlling VirtualBox. The command and stderr is shown below.

Command: ["startvm", "a6f83dd0-c3a2-4330-88c6-0aabd3cecc56", "--type", "headless"]

Stderr: VBoxManage: error: The virtual machine 'RIOT-VM' has terminated unexpectedly during startup with exit code 1 (0x1)
VBoxManage: error: Details: code NS_ERROR_FAILURE (0x80004005), component MachineWrap, interface IMachine
```

emmanuelsearch, 2017-03-27:
Anyone out there with the same issues?

thomaseichinger, 2017-03-27:
@emmanuelsearch I can't reproduce this, I had to update the VirtualBox Extension kit however too to get rid of the following error message.
```txt
There was an error while executing `VBoxManage`, a CLI used by Vagrant
for controlling VirtualBox. The command and stderr is shown below.

Command: ["startvm", "54959472-9dcb-427a-b8d5-c7f591ee57b7", "--type", "headless"]

Stderr: VBoxManage: error: Implementation of the USB 3.0 controller not found!
VBoxManage: error: Details: code NS_ERROR_FAILURE (0x80004005), component ConsoleWrap, interface IConsole
```

emmanuelsearch, 2017-03-27:
Thanks. By updating the extension pack I could make the setup work, but my (previous) VM seems to not work with the upgrade. I'll investigate further to see if I can salvage it somehow.

PeterKietzmann, 2019-02-21:
@emmanuelsearch this is almost two years old and I think it's outdated. We're currently at vbox 6.0 and vagrant 2.2.3 so I'm closing this one. Please correct me I'd I'm mistaken

closed 2019-02-21 by PeterKietzmann