RIOT-OS/RIOT #6801: Configure timer2 as counter

shreeharsha22, 2017-03-28:
Hi all,
I want to configure timer2 as counter generating interrupt every 1msec. please find the below attached program.
Issue: The problem is if I can see the variable value read after 10msec is not accurate. we are missing a tick. 
can any one tell how to avoid timer_set in  callback function to reset the timer.

[main.txt](https://github.com/RIOT-OS/RIOT/files/874809/main.txt)
[periph_conf.txt](https://github.com/RIOT-OS/RIOT/files/874810/periph_conf.txt).

Thanks in advance.

Regards,
Shreeharsha I

haukepetersen, 2017-03-28:
Hi! Could you push your changes into a branch please? This makes it much easier to see the diff of your changes... Also this kind of question is best put on the users mailing list, as this seems not to be a RIOT issue per se?! Anyway, next time...

Could you please dump the output of your test run?

kaspar030, 2017-04-25:
> Issue: The problem is if I can see the variable value read after 10msec is not accurate. we are missing a tick.

Try "timer_set_absolute()", carrying around the absolute target value.

If you have further questions on how to use RIOT, please try our mailing list (users@riot-os.org). The issue tracker is meant to be used with issues within the RIOT code.

closed 2017-04-25 by kaspar030