RIOT-OS/RIOT #6817: RTC tick value not accurate

shreeharsha22, 2017-03-29:
Hi all,
Am working on samr21-xpro board. I have configured the RTC as counter for generating an event for every one sec. For functionality testing purpose am incrementing a variable in the call back function. 
For a delay of 50 sec am expecting value of the variable to be 50 but am getting the value as 49 i,e missing one tick value. The same experiment is repeated for 100,400 secs and tick value is 99 and  399.  May I know whether is it expected behaviour or is it possible to get accurate ticks. 

Thanks in advance.
[ex.txt](https://github.com/RIOT-OS/RIOT/files/879443/ex.txt)
[main.txt](https://github.com/RIOT-OS/RIOT/files/879450/main.txt)

Regards,
Shreeharsha I

dylad, 2017-03-30:
Hello,
Is it the same issue as #6331 ?

kYc0o, 2017-04-25:
I think so, but let's keep it open for now.

miri64, 2017-10-25:
That one was fixed by https://github.com/RIOT-OS/RIOT/pull/7476 so I'm closing this as well

closed 2017-10-25 by miri64