RIOT-OS/RIOT #6830: doc: `@ingroup` tags in *.c files

miri64, 2017-03-31:
Throughout the code-base some (but not all) c-files contain a Doxygen `@ingroup` tag. First of all, shouldn't it be an `@addtogroup` tag instead (since it does not define a new group). Second, why is it there in the first place? C-files are not compiled by Doxygen.

miri64, 2017-03-31:
I would prefer to remove them, since there is no way to check if they are correct anyways (other than compiling the c-files with doxygen).

smlng, 2017-03-31:
mhm, from the doxygen docu one either uses `@defgroup` together with `@ingroup` tags **or** `@addtogroup` *instead* of the other two

miri64, 2017-03-31:
> mhm, from the doxygen docu one either uses @defgroup together with @ingroup tags or @addtogroup instead of the other two

That's what I said ;-) "First of all, shouldn't it be an @addtogroup tag instead (since it does not define a new group)."

smlng, 2017-03-31:
do we have any _rules_ when to use which variant for documentation?

miri64, 2017-03-31:
Nope.

smlng, 2017-03-31:
as expected, then ... its inconsistent all over the place, even if `@defgroup` is used its not clear where to define it first. 

For instance:

- `sys/include/hashes.h` has an `@ingroup sys_hashes`, the group is defined in `sys/hashes/doc.txt`
- `sys/include/fmt.h` has `@defgroup sys_fmt` and `@ingroup sys` in itself, no extra `doc.txt`
- `sys/include/fd.h` simply uses `@addtogroup posix`

smlng, 2017-03-31:
might be something for the DocTF, if still active?

miri64, 2017-03-31:
> might be something for the DocTF, if still active?

sporadic activeness :rofl:

miri64, 2019-08-10:
I don't care about this anymore

closed 2019-08-10 by miri64