RIOT-OS/RIOT #6840: Stack Buffer Overflow - Potential Denial of Service

JeffJerseyCow, 2017-04-01:
I'd like to report a stack buffer overflow bug within RIOT without making it publicly known. I'm still doing some testing at the device level but a bug definitely exists, who is best to discuss with privately?

lebrush, 2017-04-01:
Why should you not make it publicly available? This is an open source project and the issue can't be discussed / solved if no information about it is shared. 
In case it's an issue, once solved a patch will be publicly available to fix it. So everyone will know about the issue after all. From this point until a hypothetical private company using RIOT issues a firmware update, it's the company responsibility.

Feel free to report about it here, in the devel mailing list (https://lists.riot-os.org/mailman/listinfo/devel) or try to contact someone over IRC (#riot-os at irc.freenode.net)

gebart, 2017-04-01:
@JeffJerseyCow Right now, I don't think there is a designated software vulnerability team among the developers, but I would suggest that you contact @OlegHahm,  @kaspar030, or @EmmanuelSearch, directly via email or IRC until we have defined a security issues group, they should be able to direct you to the person most suited to address the vulnerability. 

@lebrush current industry best practice when it comes to vulnerabilities in open source projects is to contact some core developers or a security team to discuss a solution and apply that solution to the tree before publicly releasing the information about how the exploit is triggered. This is done to avoid having exploits being used in the wild before the developers have had s chance to fix them. https://en.m.wikipedia.org/wiki/Responsible_disclosure

lebrush, 2017-04-01:
It sounds sensitive and I think makes a lot of sense for systems which are easily updatable. However, it's quite hard for on-the-field embedded devices since a fix may not be even possible, if there's no update possibilities... Newly produced devices could have the issue fixed though... 

I guess that doing it the way @gebart suggests does not hurt anyone and could benefit [companies with] updatable devices on the field ;-) So between no benefit and some benefit... this is the best solution :+1: Good point @gebart !

Maybe we should document this somewhere, define contact people and define a procedure for announcing security issues (i.e. new dedicated mailing).

JeffJerseyCow, 2017-04-01:
@gebart Thank you I appreciate it. I've set out on attacking various IoT devices as a preventative measure to future security issues.
@lebrush The problem occurs when a vulnerability exists and is used maliciously before a responsible disclosure occurs; god forbid another Mirai occurs without any preventative measures.

PeterKietzmann, 2017-04-07:
security@riot-os.org has been set up and documented in the [wiki](https://github.com/RIOT-OS/RIOT/wiki/FAQ) so I close this issue. Please reopen if I'm mistaken

closed 2017-04-07 by PeterKietzmann