RIOT-OS/RIOT #6842: EMCUTE with SAMR21-XPRO device

andyrosete17, 2017-04-02:
Hi I am new to this,

I want to connect a board with the linux machine and test the MQTT protocol. I don't know how to set the IP in the board and how to communicate it with the Linux machine.
Thanks in Advance.

With Regards,
A.Rosete

smlng, 2017-04-03:
@andyrosete17, 

I'd suggest that you first get a little bit familiar with RIOT and required toolchains, for samr21-xpro you need the arm-none-eabi GCC and openocd (flash tool). The tutorials should get you started on this, see [here](https://github.com/RIOT-OS/Tutorials). 

Afterwards, you can try out the MQTT (SN) example provided with RIOT, see [here](https://github.com/RIOT-OS/RIOT/tree/master/examples/emcute).

andyrosete17, 2017-04-03:
@smlng 

Thanks for the suggestion as you said, I've already done the tutorials and I've run the protocol in local as the example explained. After that I flashed the samr21 with the mqtt (sn) example modified but i don't know how to get/set the ip address of the samr21 device, and the way of communication with the device and the linux machine  because in the example is easy using the terminal and setting the IP, but I guess that the device have to do that by his own when it is flashed. I've been looking for examples and some teoric information bu iI think I'm not searching good because I didn't found any information about it. 

Thanks in advance

smlng, 2017-04-03:
ah okay, you already past the native stuff 👍 Perfect!

The `examples/emcute` application provides a shell to run commands. On the one hand to configure the mqtt stuff (connect to broker, pub/sub topics), but on the other hand you can also get and set network parameters using `ifconfig` command.

However, the problem is that by default you have no networking between your linux host and the riot board - the usb just provides a serial terminal connection. So you either need a second SAMR21 board and flash the boarder-router example or you need a linux host with a 802.15.4 radio, e.g. a RaspberryPi with openlabs or ATUSB. You may also extend the emcute example to support ethos or something like that.

andyrosete17, 2017-04-03:
@smlng ,

Uhmm I thought i could test the protocol between the device and the linux machine. I have 5 Samr21 devices and one Raspberry pi (very old one, model B). 
So I should flash one of them with the boarder-route example (https://github.com/RIOT-OS/RIOT/wiki/Tutorial:-RIOT-and-Multi-Hop-Routing-with-RPL), but later how can I communicate the device, and in case of that, how can I be sure of the success of the communication.

andyrosete17, 2017-04-09:
Hello I've flashed two device with the emcute example but with the samr21-xpro devices I could not connect both of them. I've been seen the ifconfig results but I don't know how to connect one with the other.
Do I need something like the boarder-router  example to have a network? in case of need that, how con emcute example be tested?
[image]

kaspar030, 2017-04-25:
Closed in favour of #6956.

closed 2017-04-25 by kaspar030