RIOT-OS/RIOT #6856: Board file for STM32f105RCT6

shreeharsha22, 2017-04-04:
Hi all,
Am working on STM32f105RCT6 board. May I know if this board is already supported by RIOT or should I develop a board file for the same. 
Looking forward for your valuable reply.

Thanks in advance.

Regards,
Shreeharsha I

aabadie, 2017-04-04:
Hi @shreeharsha22,

And thanks for your interest in RIOT.

AFAIK, STM32f105RCT6 is not an exact board reference but a ST MCU reference. What is the board you are targeting exactly, do you have a link ? is it open-hardware ?

Concerning STM32f105RCT6 MCU, RIOT already supports this ST family of MCU (see [here](https://github.com/RIOT-OS/RIOT/tree/master/cpu/stm32f1) in the code for details), so adding support for it is just a matter of:
* adding the CMSIS vendor files from ST (you can get it using STM32CubeMX software) in `stm32f1/include/vendor`. You might also need to modify it slightly.
* updating `stm32f1/include/cpu_conf.h` in order to include the right cmsis header file
* adding a linker script in `stm32f1/ldscripts` for this particular MCU

Let us know if this helps.

shreeharsha22, 2017-04-04:
Hi aabadie ,
As suggested  in the previous comment I did following steps but dint get .ld file form STM32CubeMX.

1. added stm32f105xc.h file to stm32f1/include/
2. Included #include "stm32f105xc.h" in cpu_conf.h

3. not able to find *.ld file in STM32CubeMX.
can u please help me for 3rd step.
Thanks in advance,

Regards,
Shreeharsha I

aabadie, 2017-04-04:
@shreeharsha22, you have to write the linker script yourself following the MCU datasheet : copy an existing one in `stm32f1/ldscripts` and just change `LENGTH` value in the `rom` and `ram` lines.

closed 2019-09-10 by stale[bot]