RIOT-OS/RIOT #6859: C++ constructors not called for global variables on native ?

astralien3000, 2017-04-04:
Everything is in the title.

I tried the following code on native and samr21-xpro : 
```c++
#include <stdio.h>
#include <map>

std::map<int,int> test;

void* __dso_handle = 0; // required for samr21-xpro, useless for native

int main(void)
{
    puts("Hello World!");
    test[2] = 5;
    puts("test");
    return 0;
}
```

native has a segfault before printing "test"
samr21-xpro seems ok

Looks like native's startup script does not call C++ constructors for global variables before the main (or even at all ?).
The startup code for native is very hard to understand for me. I will try to investigate, but any information about this is welcome !

gebart, 2017-04-04:
I actually solved this problem today. See #6858 for a proposed solution.

astralien3000, 2017-04-04:
Thanks ! I am very lucky you fixed that before me ^^ !

gebart, 2017-04-04:
Reopening until we have merged a solution

astralien3000, 2017-04-13:
It can be closed now.

closed 2017-04-13 by astralien3000