RIOT-OS/RIOT #6865: nrf52dk status

kYc0o, 2017-04-05:
I'm currently trying to use this board for some development but I found that is broken since release 2016.07.

It is certainly due to the nordic softdevice library which was broken since some time ago, but was never fixed. Since then, several changes to RIOT can also make difficult to fix it.

So, my proposition is to disable the softdevice for this board and enable it when it gets fixed.

Any suggestions?

PeterKietzmann, 2017-04-05:
Related to #6379?

kYc0o, 2017-04-05:
Certainly. Should we close this one since it's a duplicate?

PeterKietzmann, 2017-04-05:
Maybe close one of both, don't know which one. Depends on the issue you describe here. Is you issue more generic? Just close the one you think makes more sense.

kYc0o, 2017-04-05:
I think the best idea should be to find why it's broken, open an issue describing it and close both, since the board it actually works without the softdevice enabled (as @haukepetersen did for nRF52840dk in #6863).

kb2ma, 2017-04-05:
Forgive my ignorance, but does the absence of softdevice mean the device is incapable of 6lo and IP?

haukepetersen, 2017-04-05:
@kb2ma yes and no, depends on how you look at it: when falling back to the `nrfmin` driver, the board is able to run 6lo and IPv6, but on it's own prorietary link layer, so it won't be compatible with BLE or 15.4...

kYc0o, 2017-04-25:
So I propose to close this issue and move into #6379.

closed 2017-04-25 by kYc0o