RIOT-OS/RIOT #6869: xbee does not work

kyungtaeH, 2017-04-07:
The xbee(tests/driver_xbee) does not working properly in the latest version.
For reference, I have confirmed that it works well in past versions.

gebart, 2017-04-07:
Could you tell us what part is not working?
What version did it work properly? (git commit id or a release number) 
When you say current version, do you mean the latest master branch or do you mean the release 2017.01?

kyungtaeH, 2017-04-07:
I just tested it with 'git clone' few days ago.
It seems that it did not work properly because I did not 'git checkout' to the release version of 2017.01. 
After checkout to release version, it works! 
I'm really sorry to report the wrong issue.

gebart, 2017-04-07:
What is not working with xbee on master?
If it is broken on master then someone needs to fix it.

kyungtaeH, 2017-04-07:
I just checked 'git pull' again from the master and it works too!
Perhaps I have checked out a version that is not temporarily available.

gebart, 2017-04-07:
ok, closing this. Post a new issue if you find anything is broken.

closed 2017-04-07 by gebart