RIOT-OS/RIOT #6888: netdev: How to report transmit statistics

bergzand, 2017-04-11:
For ETX calculations, it is useful to know how many transmissions the radio had to perform before an ACK was received. Currently the netdev/driver glue code does not seem to have a way to report these values back to netdev. Only success or failure is reported.

Is there a preferred way to implement this? I'm willing to write the code for this, but I have no clue what kind of structure is wanted for this.

miri64, 2017-04-11:
If the value is on a device level and changes seldom a new NETOPT would be the preferred way to report this.

If the value is on a per packet basis, I have a little history lesson ;-): When the `recv()` function got the info context, the `send()` was supposed to get one too. However, we decided [to provide it when it is needed](https://github.com/RIOT-OS/RIOT/pull/4648#issuecomment-177467596). So I guess the time would be now ;-).

bergzand, 2017-04-11:
This is indeed about values on a per packet basis.

The problem with giving the `send()` method an info context is that the transmission statistics are known when the callback function is called. Would it be okay to add an info context to the netdev event callback function, [here](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/link_layer/netdev/gnrc_netdev.c#L49)?

haukepetersen, 2017-04-25:
Does it make sense to go the same path than we did for `recv` by changing the send function to something like 
```c
int (*send)(netdev_t *dev, const struct iovec *vector, unsigned count, void *info);
```
and define something similar to `struct netdev_radio_rx_info` like
```c
struct netdev_radio_tx_info {
    uint8_t retransmits;
}
```
?

bergzand, 2017-05-05:
@haukepetersen I think that extending the send function doesn't work here. Most if not all send functions initiate a send operation in the hardware radio. The result of this operation is not available until the radio finished the transmission, but the send function has already returned by then.
Because of this asynchronous working, I proposed to add the information struct somewhere in the callback functions.

miri64, 2017-05-05:
> Because of this asynchronous working, I proposed to add the information struct somewhere in the callback functions.

Why not use the TX_COMPLETE event then?

bergzand, 2017-05-05:
As far as I can tell, the TX_COMPLETE event only indicates that a transmission was successful, not how "hard" the radio had to try to get it to the other node. 
The problem with this is that with ACK request enable, an ieee802.15.4 transmitter should retry 3 times if it doesn't receive an ACK. So for example, a single send call could have a radio transmit 3 times before an ACK from the receiving node was received. The current code would only indicate a TX_COMPLETE for this without any indication that this operation had cost 3 actual transmissions.

For ETX calculations, information about how many times a transceiver had to send a transmission before an ACK was received is critical information. In the future I would like to use ETX for path selection in routing protocols and there some indication of the actual link quality is quite useful there.

haukepetersen, 2017-05-11:
> I think that extending the send function doesn't work here

Yes, you are right.

Thinking about this, could a combination of something like `netdev->get(NETOPT_TX_RETRIES_NEEDED)` and the TX_COMPLETE be the solution here? The number of retries would need to be counted by the driver internally and the value must of course be synchronized with the get call, but this could be simply done by having a shadow value that does the acutal count and an exposed value that is showing the count for the last successful transmit.

bergzand, 2017-08-02:
Finally some time to work on this again.

Your suggestion sounds better than adding extra void pointers to the callback :)
I'll see if I can extend the netopts and add this functionality in #6873.

bergzand, 2017-10-12:
Fixed with #7447, #7484 and #7448

closed 2017-10-12 by bergzand