RIOT-OS/RIOT #6896: C standard forced on platforms

astralien3000, 2017-04-13:
I have hard time to use external c11 code with RIOT on some platforms, because `-std=gnu99` is forced in [boards' Makefile.include (native, for example)](https://github.com/RIOT-OS/RIOT/blob/master/boards/native/Makefile.include#L49).

Is there a reason to force this, an not to do like in [this file](https://github.com/RIOT-OS/RIOT/blob/master/makefiles/cflags.inc.mk#L1-L7) ?

kaspar030, 2017-04-13:
> Is there a reason to force this

There's no reason to force it. The default has to be overwritten, though, as those boards need the gnu extensions.

astralien3000, 2017-04-14:
Ok, so we could enable the flag to be overwritten by the user ?
But I am lost in the Makefiles : 
 - There is one in [boards](https://github.com/RIOT-OS/RIOT/blob/master/boards/native/Makefile.include)
 - and 2 in `makefiles/arch` ([msp430](https://github.com/RIOT-OS/RIOT/blob/master/makefiles/arch/msp430.inc.mk), [mips](https://github.com/RIOT-OS/RIOT/blob/master/makefiles/arch/mips.inc.mk))

closed 2017-10-13 by astralien3000

aabadie, 2017-10-13:
@astralien3000, may I ask why you are closing ? Did you find a solution ?