RIOT-OS/RIOT #6910: doc: HTML docs broken with Doxygen >=1.8.12

gebart, 2017-04-14:
https://github.com/doxygen/doxygen/commit/8480d35beef57ed08139b58972bfb83a3b37422c
seem to break the new design from #6878 
Screenshot:
[image]

Web browser console shows:
```
Uncaught TypeError: $(...).smartmenus is not a function
    at initMenu (menu.js:25)
    at HTMLDocument.<anonymous> (modules.html:83)
    at i (jquery.min.js:2)
    at Object.fireWith [as resolveWith] (jquery.min.js:2)
    at Function.ready (jquery.min.js:2)
    at HTMLDocument.K (jquery.min.js:2)
```

miri64, 2017-04-14:
Can you send me the html directory zipped?

gebart, 2017-04-14:
Try this link: https://www.dropbox.com/sh/8elyl6kb6at1j9k/AABuCBA7XS7DZDOTLhvjXbQwa?dl=0

miri64, 2017-04-14:
Ah okay, the use this one https://github.com/vadikom/smartmenus now for the menus instead of handcrafting them. I will look into that. For now since this isn't in the auto-gernerated doc, let's keep this issue open.

miri64, 2017-04-15:
@gebart can you try with https://github.com/RIOT-OS/RIOT/pull/6914?

closed 2017-04-25 by gebart

axelosorio, 2018-09-21:
this solution work for me
https://www.smartmenus.org/forums/topic/conflicting-jquery/