RIOT-OS/RIOT #6930: Potential compilation error

kamejoko80, 2017-04-19:
Hello,

I see in the source code there is some potential compilation error.

1> Inline function definition: 
In file /drivers/kw2xrf/include/kw2xrf_intern.h
Inline function in the header file should be defined with "static inline". 
Compilation error will happen if we change the gcc flag from -Os to -O0

2> Partially include source file in nanocoap package:
In file /pkg/nanocoap/Makefile.nanocoap 
handler.c should be included into Makefile.nanocoap. The is a function coap_handle_req() in nanocoap.c references to coap_resources[] which is defined in handler.c. 
However coap_hanlde_req() is not called, gcc can compile successfully because of the option flag -ffunction-sections -fdata-sections.

smlng, 2017-04-24:
I think you're right, but how did you *see* these potential errors? Any specific tool you used, or example code so that others can reproduce your findings?

As these changes look rather minor: can you provide *separate* PRs for these issues?

smlng, 2017-07-06:
closing, because (1) is resolved all inline functions are also static now (see #7108) and (2) is no issue because `coap_resources` is statically defined by the application and passed to gcoap.

closed 2017-07-06 by smlng