RIOT-OS/RIOT #6950: question regarding best practice for sizeof() type

kaspar030, 2017-04-24:
While experimenting with coccinelle, I realized that we're not using sizeof() consistently regarding to using a type or an identifier as argument.

For example, grep for usages of ```gnrc_netapi_get()```.

Some occurences are using ```gnrc_netapi_get(..., &var_name, sizeof(var_name))```, some are using ```gnrc_netapi_get(..., &var_name, sizeof(var_type))```.

Should we aim for consistency here? If yes, which way?

gebart, 2017-04-24:
I prefer to use `sizeof` on the variable itself, when possible, because then we don't need to search and update all sizeof uses if the type of a variable changes. Drawback: This approach does not work if the variable is of a pointer type and the target is an array and not a single value. (`sizeof(*ptr)` is the size of one item, does not take into account the array size)

kaspar030, 2019-08-19:
unconclusive, nit sure it matters. closing for now.

closed 2019-08-19 by kaspar030