RIOT-OS/RIOT #6956: Communicate two samr21-xpro Emcute example and gnrc_border_route example

andyrosete17, 2017-04-24:
Hi

I'm trying to communicate two devices (samr21-xpro) one of them with emcute (mqtt-sn) example and the gnrc_border_route example, but the con command of emcute example return TimeOut error. 

I've added the ip of the gnrc_border_route example to emcute example

[image]    
(gnrc_border_route)    

[image]
(emcute with the ip added)

if its executed ping6 command it can receive a response of gnrc example
[image]

But when the con command is executed it return a TimeOut error (res= -4) (it say successfully connected but it must be a bug)
[image]

After fill the code with debug lines i saw that the line that fail is 
thread_t *me = (thread_t*) sched_active_thread;

this code segment for example 

emcute.c
 static int syncsend(uint8_t resp, size_t len, bool unlock)
{
    int res = EMCUTE_TIMEOUT;
    waiton = resp;
    timer.arg = (void *)sched_active_thread;
    /* clear flags, in case the timer was triggered last time right before the
     * remove was called */
    thread_flags_clear(TFLAGS_ANY);
    for (unsigned retries = 0; retries < EMCUTE_N_RETRY; retries++) {
        sock_udp_send(&sock, tbuf, len, &gateway);
        xtimer_set(&timer, (EMCUTE_T_RETRY * US_PER_SEC));
        thread_flags_t flags = thread_flags_wait_any(TFLAGS_ANY);
        if (flags & TFLAGS_RESP) {
            xtimer_remove(&timer);
            res = result;
            retries = EMCUTE_N_RETRY;
        }

    }
     
    /* cleanup sync state */
    waiton = 0xff;
    if (unlock) {
        mutex_unlock(&txlock);
    }
    DEBUG("The final result = %d\n", res);
    return res;
}

--------------------------------------------------------------------------------------------------

thread_flag.c
thread_flags_t thread_flags_wait_any(thread_flags_t mask)
{
    thread_t *me = (thread_t*) sched_active_thread;
    _thread_flags_wait_any(mask);
    thread_flags_t result = _thread_flags_clear_atomic(me, mask);
    return result;
}

because it try to inittialize the thread pointer but the flag is always different
 if (flags & TFLAGS_RESP)

The values are:
The flag = 0x00000002 (This mean TIMEOUT) and TFLAGS_RESP = 0x00000001

Please is somebody know what do I need to do, I will appreciate a lot an advice.

andyrosete17, 2017-05-01:
Hi, 

I'm still having problem to connect the devices  and establish the communication among then.  I have been testing the emcute example in native and it run perfectly but once I want to use the samr21  it fails.  I've been trying to test the emcute example with the gnrc_border_router example but I'm not able to communicate them even when I've added the ip and I can execute ping6 and it response is positive. 
Please if somebody know how can I resolve this,  I will appreciate a lot the help. 

Thanks in advance...

haukepetersen, 2017-10-25:
Sorry for the (very) late reply. 

Just tested your setup also using two `samr21-xpro` nodes: it worked like a charm. Which mqtt broker did you try to connect to? Did you start one as described in the README of the `emcute` example? And did you connect to the correct address and port of that mqtt-sn broker?

The `success` message after a connection timeout is indeed an error, fix is here: #7848

As the example is working here (except for the minor bug fix linked above), I consider this issue solved. Please shout/re-open if needed!

closed 2017-10-25 by haukepetersen