RIOT-OS/RIOT #6967: Questions regarding support for STM32F103C8

nbars, 2017-04-25:
Hey guys,

I just tried to compile RIOT for an STM32F103C8 device and noticed that it doesn't seem to be supported. Can you tell me what I have to do for adding support for this chip?

Am I right in the assumption that I only need to change the file ```/cpu/stm32f1/include/cpu_conf.h``` and get somewhere a vendor file for this chip?
Or are there any other places that need to be edited?

Thanks in advance

kamejoko80, 2017-06-14:
Hello,

You can refer board nucleo-f103 and port to your new board.

smlng, 2017-10-20:
@nbars have a look at #7727, that should work - assuming you've one of those very cheap stm32f103c8 boards (from eBay for instance).

smlng, 2017-10-27:
closing with #7727 merged

closed 2017-10-27 by smlng