RIOT-OS/RIOT #6973: Troubles with IPv6 autoconfiguration when using XBee

aabadie, 2017-04-26:
Hi,

I have issues with IPv6 address autoconfiguration on a XBee module: it never gets the prefix propagated by a border router (I'm using a Raspberry with OpenLabs as BR). The interface is only configured with a link local address.
The problem seems to be there for a long time now: it's working with 2016.10 but is broken in 2017.01. I suspect #6235 to have introduced this regression but a bisect is required to be sure of that.

My node is an Arduino-zero (SAMD21) flashed with an adapted `gnrc_networking` example (see #6971).
I also enabled debug in xbee.c to have a bit more information and I can see something strange with the channel setup. Here is the output when rebooting my node:
```
2017-04-26 13:10:14,011 - INFO # [xbee] AT_CMD: +++
2017-04-26 13:10:15,057 - INFO # [xbee] AT_CMD: ATMM2
2017-04-26 13:10:15,063 - INFO # [xbee] AT_CMD: ATAP1
2017-04-26 13:10:15,069 - INFO # [xbee] AT_CMD: ATAC
2017-04-26 13:10:15,074 - INFO # [xbee] AT_CMD: ATCN
2017-04-26 13:10:15,079 - INFO # [xbee] AT_CMD: SH
2017-04-26 13:10:15,104 - INFO # [xbee] AT_CMD: SL
2017-04-26 13:10:15,128 - INFO # [xbee] AT_CMD: CH      // weird character here
2017-04-26 13:10:15,149 - INFO # [xbee] AT_CMD: ID
2017-04-26 13:10:15,172 - INFO # [xbee] init: Initialization successful
2017-04-26 13:10:15,176 - INFO # [xbee] send: now sending out 58 byte
2017-04-26 13:10:15,242 - INFO # main(): This is RIOT! (Version: 2017.04-devel-886-g524ba8-trek-saclay)
2017-04-26 13:10:15,245 - INFO # RIOT network stack example application
2017-04-26 13:10:15,248 - INFO # All up, running the shell now
```

`ifconfig` shows that there's no channel set (could it be because of the response timeout ?) :
```
> ifconfig
2017-04-26 13:10:33,079 - INFO #  ifconfig
2017-04-26 13:10:33,083 - INFO # Iface  7   HWaddr: c2:4c [xbee] AT_CMD: CH
2017-04-26 13:10:34,093 - INFO # [xbee] api_at_cmd: response timeout
2017-04-26 13:10:34,095 - INFO # [xbee] AT_CMD: ID
2017-04-26 13:10:34,116 - INFO #  NID: 0x23
2017-04-26 13:10:34,120 - INFO #            Long HWaddr: 00:13:a2:00:41:46:c2:4c 
2017-04-26 13:10:34,121 - INFO #           
2017-04-26 13:10:34,124 - INFO #            MTU:1280  HL:64  
2017-04-26 13:10:34,127 - INFO #            Source address length: 8
2017-04-26 13:10:34,130 - INFO #            Link type: wireless
2017-04-26 13:10:34,135 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2017-04-26 13:10:34,141 - INFO #            inet6 addr: fe80::213:a200:4146:c24c/64  scope: local
2017-04-26 13:10:34,147 - INFO #            inet6 addr: ff02::1:ff46:c24c/128  scope: local [multicast]
2017-04-26 13:10:34,152 - INFO #            inet6 addr: ff02::1a/128  scope: local [multicast]
2017-04-26 13:10:34,153 - INFO #            
2017-04-26 13:10:34,158 - INFO #            Protocol or device doesn't provide statistics.
2017-04-26 13:10:34,161 - INFO #            Statistics for IPv6
2017-04-26 13:10:34,164 - INFO #             RX packets 0  bytes 0
2017-04-26 13:10:34,168 - INFO #             TX packets 1 (Multicast: 1)  bytes 50
2017-04-26 13:10:34,171 - INFO #             TX succeeded 1 errors 0
```

I also tried to ping the border router. The first packet times out but the 2 after works:
```
2017-04-26 13:34:32,988 - INFO #  ping6 fe80::1ac0:ffee:1ac0:ffee%7
2017-04-26 13:34:32,991 - INFO # [xbee] send: now sending out 66 byte
2017-04-26 13:34:34,061 - INFO # ping timeout
2017-04-26 13:34:35,065 - INFO # [xbee] send: now sending out 66 byte
2017-04-26 13:34:35,185 - INFO # [xbee] isr: data available, waiting for read
2017-04-26 13:34:35,190 - INFO # [xbee] recv: reading size without dropping: 29
2017-04-26 13:34:35,194 - INFO # [xbee] recv: consuming packet: reading 29 byte
2017-04-26 13:34:35,201 - INFO # 12 bytes from fe80::1ac0:ffee:1ac0:ffee: id=83 seq=2 hop limit=255 time = 132.165 ms
2017-04-26 13:34:36,206 - INFO # [xbee] send: now sending out 66 byte
2017-04-26 13:34:36,326 - INFO # [xbee] isr: data available, waiting for read
2017-04-26 13:34:36,330 - INFO # [xbee] recv: reading size without dropping: 29
2017-04-26 13:34:36,334 - INFO # [xbee] recv: consuming packet: reading 29 byte
2017-04-26 13:34:36,342 - INFO # 12 bytes from fe80::1ac0:ffee:1ac0:ffee: id=83 seq=3 hop limit=255 time = 132.132 ms
2017-04-26 13:34:36,346 - INFO # --- fe80::1ac0:ffee:1ac0:ffee ping statistics ---
2017-04-26 13:34:36,352 - INFO # 3 packets transmitted, 2 received, 34% packet loss, time 3.06351821 s
```

And after that the xbee starts to receive packet but still no prefix.

Any idea ?

aabadie, 2017-04-26:
@miri64, @haukepetersen do you have any idea about this problem ? It would be great to fix it for the release (4 days remaining!).

miri64, 2017-04-26:
I just tried this on Arduino Due and xbee and I get the prefix, but I ping to the global address of the border router times out (link-local works though). I can sniff the replies though.

aabadie, 2017-04-26:
> I just tried this on Arduino Due and xbee and I get the prefix, but I ping to the global address of the border router times out

What is your prefix ? I'm trying with fd00:aaaa::/64. What kind of BR do you have ?

aabadie, 2017-04-26:
@miri64 I tested with arduino-due and it doesn't work either but maybe I didn't setup the shield correctly. What is your setup ?

miri64, 2017-04-26:
The one @haukepetersen handed me a few weeks back: at UART1 (Ports 18 and 19) and with the parameters you provided in #6971:

[image]

aabadie, 2017-04-26:
Ok I see, I had the same. Tested the shield back on arduino-zero and I get a link-local address.
Here is my output at reboot with arduino-due:
```
2017-04-26 18:17:28,354 - INFO #  reboot
2017-04-26 18:17:29,457 - INFO # [xbee] AT_CMD: +++
2017-04-26 18:17:30,558 - INFO # [xbee] AT_CMD: ATMM2
2017-04-26 18:17:30,566 - INFO # [xbee] AT_CMD: ATAP1
2017-04-26 18:17:30,571 - INFO # [xbee] AT_CMD: ATAC
2017-04-26 18:17:30,575 - INFO # [xbee] AT_CMD: ATCN
2017-04-26 18:17:30,583 - INFO # [xbee] AT_CMD: SH 
2017-04-26 18:17:30,583 - INFO # 
2017-04-26 18:17:31,591 - INFO # [xbee] api_at_cmd: response timeout
2017-04-26 18:17:31,595 - INFO # [xbee] init: error getting address
2017-04-26 18:17:31,599 - INFO # [xbee] send: now sending out 58 byte
2017-04-26 18:17:31,665 - INFO # main(): This is RIOT! (Version: 2017.04-devel-887-g804e-trek-saclay-examples_xbee_config)
2017-04-26 18:17:31,669 - INFO # RIOT network stack example application
2017-04-26 18:17:31,672 - INFO # All up, running the shell now
```

haukepetersen, 2017-05-02:
just tried this using an xbee @ arduino-due and a samr21-xpro/gnrc_border_router. My arduino gets a global prefix from the border router just fine:
```
2017-05-02 14:23:17,973 - INFO # ifconfig
2017-05-02 14:23:17,977 - INFO # Iface  7   HWaddr: fc:07 [xbee] AT_CMD: CH 
2017-05-02 14:23:17,998 - INFO #  Channel: 26 [xbee] AT_CMD: ID 
2017-05-02 14:23:18,018 - INFO #  NID: 0x23
2017-05-02 14:23:18,023 - INFO #            Long HWaddr: 00:13:a2:00:40:a9:fc:07 
2017-05-02 14:23:18,023 - INFO #           
2017-05-02 14:23:18,027 - INFO #            MTU:1280  HL:64  RTR_ADV  
2017-05-02 14:23:18,031 - INFO #            Source address length: 8
2017-05-02 14:23:18,035 - INFO #            Link type: wireless
2017-05-02 14:23:18,039 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2017-05-02 14:23:18,043 - INFO #            inet6 addr: fe80::213:a200:40a9:fc07/64  scope: local
2017-05-02 14:23:18,051 - INFO #            inet6 addr: ff02::1:ffa9:fc07/128  scope: local [multicast]
2017-05-02 14:23:18,055 - INFO #            inet6 addr: ff02::1a/128  scope: local [multicast]
2017-05-02 14:23:18,060 - INFO #            inet6 addr: 2001:db8::213:a200:40a9:fc07/64  scope: global
2017-05-02 14:23:18,067 - INFO #            inet6 addr: ff02::2/128  scope: local [multicast]
2017-05-02 14:23:18,068 - INFO #            
2017-05-02 14:23:18,072 - INFO #            Protocol or device doesn't provide statistics.
2017-05-02 14:23:18,076 - INFO #            Statistics for IPv6
2017-05-02 14:23:18,080 - INFO #             RX packets 12  bytes 996
2017-05-02 14:23:18,084 - INFO #             TX packets 14 (Multicast: 4)  bytes 790
2017-05-02 14:23:18,087 - INFO #             TX succeeded 14 errors 0
2017-05-02 14:23:18,088 - INFO # 
```
But I think I might have found some small things with the xbee driver, will propose fixes soon.

haukepetersen, 2017-05-02:
oh, and the output looks a little weird, as I have DEBUG for the xbee driver turned on currently...

aabadie, 2017-05-04:
I also has debug turned on in the xbee driver but that didn't change anything.
The only difference with my setup is that you are using RIOT BR and I'm using a RPI + OpenLabs BR (that works like a charm with samr21 and zolertia remote).
Will test with a RIOT BR.
I still don't understand why my Arduino due + Xbee doesn't work.

Do you have a Nucleo 144 around ? I am interested to see if you also can make it work on it.

haukepetersen, 2017-05-05:
I have the nucleo144-f746 I can try it with.

haukepetersen, 2017-05-05:
done: using the `gnrc_networking` with `xbee` @ `nucleo144-f746` works (almost) as expected, I get identical behavior as when using the same configuration on the `arduino-due` (this time without `DEBUG` for the driver enabled). Initialization goes fine and a prefix from the border router is assigned (again using a `samr21-xpro` with the `gnrc_border_router` example):
```
2017-05-05 13:41:50,466 - INFO #  ifconfig
2017-05-05 13:41:50,510 - INFO # Iface  7   HWaddr: fc:07  Channel: 26  NID: 0x23
2017-05-05 13:41:50,514 - INFO #            Long HWaddr: 00:13:a2:00:40:a9:fc:07 
2017-05-05 13:41:50,515 - INFO #           
2017-05-05 13:41:50,518 - INFO #            MTU:1280  HL:64  RTR_ADV  
2017-05-05 13:41:50,522 - INFO #            Source address length: 8
2017-05-05 13:41:50,524 - INFO #            Link type: wireless
2017-05-05 13:41:50,530 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2017-05-05 13:41:50,535 - INFO #            inet6 addr: fe80::213:a200:40a9:fc07/64  scope: local
2017-05-05 13:41:50,542 - INFO #            inet6 addr: ff02::1:ffa9:fc07/128  scope: local [multicast]
2017-05-05 13:41:50,547 - INFO #            inet6 addr: ff02::1a/128  scope: local [multicast]
2017-05-05 13:41:50,553 - INFO #            inet6 addr: 2001:db8::213:a200:40a9:fc07/64  scope: global
2017-05-05 13:41:50,558 - INFO #            inet6 addr: ff02::2/128  scope: local [multicast]
2017-05-05 13:41:50,560 - INFO #            
2017-05-05 13:41:50,565 - INFO #            Protocol or device doesn't provide statistics.
2017-05-05 13:41:50,567 - INFO #            Statistics for IPv6
2017-05-05 13:41:50,570 - INFO #             RX packets 2  bytes 256
2017-05-05 13:41:50,575 - INFO #             TX packets 1 (Multicast: 1)  bytes 50
2017-05-05 13:41:50,578 - INFO #             TX succeeded 1 errors 0
```

The only thing that I noticed: when calling `ifconfig` for the first time after a reboot, the driver gets an `response timeout` when trying to read the current channel. All successive `ifconfig` calls don't show that behavior, though. From the outside this looks like a variable initialization issue in the driver, but I couldn't locate it in the code (yet).

aabadie, 2017-05-19:
@haukepetersen, I just gave another try to this (using my freshly installed toolchain ;)). Here is what I have now:
* it works when using a samr21-xpro or iotlab-m3 node as BR
* it doesn't work when using Linux + OpenLabs as BR. An iotlab-m3 or a samr21-xpro gets a prefix from this BR

haukepetersen, 2017-05-22:
This is strange behavior indeed. But I fear I won't have much time the next weeks to look into this issue any further -> @miri64: could you help @aabadie to find the source of this?

miri64, 2017-05-22:
After the Ascension weekend I'll definitely can look into this (if you provide me the hardware). The next three days I don't have time for that either.

aabadie, 2017-06-29:
> After the Ascension weekend I'll definitely can look into this

Any chance for looking into this ?

miri64, 2017-06-29:
Not today anymore :-/. I'm not sure I will find the time tomorrow. But I'll try.

aabadie, 2017-07-04:
@miri64 could you find some time to look into this one ?

miri64, 2017-07-04:
I only have tomorrow and the day after, after that I'm on vacation until IETF. I see if I find the Xbee tomorrow, but I can't promise much :-/

miri64, 2017-07-05:
Mhhh.... for some reason the xbee doesn't send a Router Solicitation (which is why the Raspberry Pi never sends an RA. The `gnrc_border_router` example sends multicast RA because it get's a prefix freshly assigned that's why the xbee gets a prefix with it there).

miri64, 2017-07-05:
@aabadie see #7319.

aabadie, 2017-07-05:
Very nice, thanks! Will test on friday.

miri64, 2017-07-05:
Then I'm already on vacation ;-)

miri64, 2017-07-05:
If you want to release on the 14th and if you want this in the release this has (and its backport) need to be merged today or tomorrow, after that I will only return to a keyboard on July 15th ;-).

kYc0o, 2017-07-05:
I'll look at it today.

aabadie, 2017-07-05:
> I'll look at it today.

You need to test that the XBee gets a prefix in 2 cases:
* with a samr21-xpro or iotlab-m3 as BR
* with a RPI (OpenLabs) as BR

For the second case, you can try next to my office (channel 11) if don't have the RPI.

kYc0o, 2017-07-05:
#7319 fixes this 😃  thanks @miri64 !

PR #7319 (2017-07-05): xbee: provide NETOPT_PROTO
Fixes #6973.

The current 6Lo-ND implementation relies on `NETOPT_PROTO` being
returned for an IEEE 802.15.4 device. This isn't ideal, since it mixes
GNRC into device code (and the next iteration of NDP will do it
better), but for now this is the only way to fix #6973.
commit 34d9c67d (2017-07-05): xbee: provide NETOPT_PROTO

Fixes #6973.

The current 6Lo-ND implementation relies on `NETOPT_PROTO` being
returned for an IEEE 802.15.4 device. This isn't ideal, since it mixes
GNRC into device code (and the next iteration of NDP will do it
better), but for now this is the only way to fix #6973.
commit e3465732 (2017-07-05): xbee: provide NETOPT_PROTO

Fixes #6973.

The current 6Lo-ND implementation relies on `NETOPT_PROTO` being
returned for an IEEE 802.15.4 device. This isn't ideal, since it mixes
GNRC into device code (and the next iteration of NDP will do it
better), but for now this is the only way to fix #6973.
closed 2017-07-05 by kYc0o