RIOT-OS/RIOT #6979: frdm-k64f: kernel panic with any example using shell

kYc0o, 2017-04-27:
While testing some examples on the frdm-k64f I realised that any example using the shell leads to a kernel panic when any command is sent. I suspect is related to the UART input buffer.

@gebart do you have a board to test?

For info: The examples seem to work on 2017.01-branch

gebart, 2017-04-28:
I don't have a frdm-k64f board to test on. I can't find anything that's off between the frdm-k64f and the mulle, which is working fine. According to @haukepetersen the pba-d-01-kw2x is also working fine.
@kYc0o Could you try to single step it through a debugger and see where it faults and post a backtrace?
I will post a PR for FRDM-K22F in a few days, which is very similar to the FRDM-K64F, but I am still waiting for an order of that hardware.

kYc0o, 2017-04-28:
Fix proposed in #6984

kYc0o, 2017-05-01:
Fixed in #6984

closed 2017-05-01 by kYc0o