RIOT-OS/RIOT #6985: Segger GDB with TUI has bad newlines.

kbumsik, 2017-05-01:
Hello, I am porting Arduino MKR1000.
I am currently flashing/debugging the device using Segger EDU connected via SWD.

To run the debugging session, I run `TUI=1 BOARD=arduino-mkr1000 make debug` (I made a new board directory for MKR1000.) And the terminal shows me this screen:

[image]

Did anybody have this problem before? I posted this question because I am not sure where to ask... The programming environment is Ubuntu GNOME3 16.04.2 LTS and the terminal is GNOME terminal.

aabadie, 2017-05-01:
Hi @kbumsik,

Good to hear that someone else is interested in mkr1000. Some times ago, I opened #6666 to add support for this board.
Maybe you can have a look/review it ?
My port doesn't use the SWD programming but [Bossa](https://github.com/shumatech/BOSSA) like in Arduino IDE: so I don't have debugging facility with it.

kbumsik, 2017-05-01:
Hi @aabadie ,
Great. I am happy that someone is working on it. I quickly looked through your PR, and it looks like there are some codes that I could add :). I will make a PR to your branch when I finish reviewing it.

aabadie, 2017-05-01:
> I will make a PR to your branch when I finish reviewing it

That would be nice !
do you also plan to add the wifi driver ?

kbumsik, 2017-05-01:
@aabadie 
Yes. I already have experience in porting ATWINC1500's wifi driver on a bare-metal MCU. But I worried it won't be compatible with RIOT's network stack...

aabadie, 2017-05-02:
@kbumsik, if you can first provide the driver on its own PR with a usable API it would be great and mergeable fast. We'll see later how to integrate it in RIOT network stack.

kbumsik, 2017-05-03:
I found a perfect alternative to GDB TUI debugging. Segger provides [Ozone](https://www.segger.com/ozone.html) a perfect GUI debugging program. It doesn't require complicated configuration. Just put an `.elf` file and it works! Closing the issue.

closed 2017-05-03 by kbumsik