RIOT-OS/RIOT #7001: Better document how to use devices that support auto_init

aabadie, 2017-05-03:
As raised in #6982, we need to better explain newcomers how to build applications using the auto_init modules.
This is the case for `examples/default` and for the networking specific examples (gnrc_networking, microcoap, nanocoap, gcoap, etc).

The idea is to create a common doxygen page explaining the principles of auto_init and listing all devices that support it. Then in the README files of concerned examples, add a link to this page.

miri64, 2017-05-03:
@aabadie could you maybe write-up a page? Just have a look at the *.md files in the `doc/doxygen/src/` directory for a template (not the mainpage.md though ;-)) and add it to the [`INPUT` configuration variable](https://github.com/RIOT-OS/RIOT/releases)

aabadie, 2017-05-04:
> @aabadie could you maybe write-up a page?

I'll see what I can do

PR #9328 (2018-06-19): sys/auto_init: improve documentation

### Contribution description

This PR is a follow-up of #9264 and fixes #7001. It's not adding a new dedicated markdown page but just improving the actual `auto_init` module doxygen documentation.

### Issues/PRs references

fixes #7001, related to #9264
closed 2018-06-19 by aabadie