RIOT-OS/RIOT #7008: Error message from SAUL_REG

Hyungsin, 2017-05-04:
I'm working on making some sensor drivers interoperable with SAUL abstraction. 

Meanwhile, I found an issue with SAUL_REG_READ function [1]. 
It returns the dimension of the output value (e.g., magnetic field -> 3) and reports an error "ENODEV" if a user tries to read from a non-existing sensor. 

But even when a sensor is normally initialized, its driver sometimes fails to provide a reading due to various reasons, such as I2C failure. To be a better interface, I think SAUL_REG_READ needs to report the reading failure to users. A simple idea is to modify each SAUL driver returns 0 if its sensor fails to read a value, given that the dimension is expected to be a positive value in a normal operation.

If it makes sense, I can create a pull request about this issue and upload some changed drivers.
 
[1] https://github.com/RIOT-OS/RIOT/blob/master/sys/saul_reg/saul_reg.c#L117

haukepetersen, 2017-05-05:
Actually, the interface already defines that it returns `-ECANCELED` in case the device failed to produce the requested data. So the SAUL read function implemented by the device driver can simply return that value in case of error...

Hyungsin, 2017-05-09:
Thanks for feedback. 
Then, I also think that device drivers should be changed so that the saul read function can return -ECANCELED when reading is failed.

haukepetersen, 2017-05-11:
Yes, if they return an error the `_ECANCELED` should be propagated. But many drivers do not return error values on their read/write functions for a simple reason: the drivers test their connectivity etc. once during initialization and expect their connection (e.g. SPI, UART...) to be working from here on. So during their read/write this connection is expected to be working and won't return any errors anyway. This design decision is based on the tradeoff between fine-grained error checking and code size efficiency...

immesys, 2017-05-12:
I think it is essential for sensor drivers to have the ability to report errors on every reading. Our deployments have motes initialize once, and then run for >5 years without rebooting. The probability of never hitting a spurious error (either in the i2c bus or in the sensor) within that time is very low. If you are lucky the error is a one-off but sometimes it results in a loss of sync with the sensor. Ideally the error would be reported, and the application can react accordingly (we prefer to reset the whole system, or abstain from kicking the watchdog).

In particular, we use the HDC1080 which is supported by the HDC1000 driver. It used to return errors if i2c_write_byte failed or i2c_acquire failed etc. Now (as of 976cf30281e which I believe was yours) it just ignores all errors. It doesn't even check if i2c operations succeed at all. Even in the few months since we reflashed with that change we have had devices returning invalid readings silently.

I assumed that this change was deliberate because of policy in SAUL (which I am not familiar with) which is why @Hyungsin took a look into it. If SAUL does support error codes, why did you remove all the error codes in the HDC1000 driver?

haukepetersen, 2017-05-12:
Yep, cutting down the `hdc1000` driver was me... In retrospect it seems like I was a little bit too enthusiastic removing the error checking. The reasons I did so back then were (a) the assumption that the connection to the device would be stable after initialization and (b) optimization in regard to code size. Assumption (a) was mostly driven from experiences with SPI and UART. 

As it seems to me now, this was a wrong assumption, and we should indeed re-introduce the error checking for I2C based device driver like the `hdc1000` (see #7049 as a starter). I am actually not sure anymore, why I was so 'optimistic' back then (I guess some part was the satisfaction of reducing code size...), so sorry for making things worse!

immesys, 2017-05-12:
Thanks for responding so quickly. No hard feelings on making things worse for a bit, these things happen in any active project with a large and diverse user base, it's hard to keep track of what is important to whom.

@Hyungsin given that the hdc1000 is fixed, and we know the pattern for reporting errors through SAUL in future, I think we can close this.

Hyungsin, 2017-05-13:
Good I will close this. Thank @immesys  and @haukepetersen  ! :)

closed 2017-05-13 by Hyungsin