RIOT-OS/RIOT #7010: fatal error: stdio.h: No such file or directory , compile error stm32f4 on RIOT

tuanle311, 2017-05-05:
I got this issue when i compile hello world for board stm32f4 discovery, RIOT

tuanle@TuanLe ~/RIOT/examples/hello-world $ BOARD=stm32f4discovery make
Building application "hello-world" for "stm32f4discovery" with MCU "stm32f4".

/home/tuanle/RIOT/examples/hello-world/main.c:23:19: fatal error: stdio.h: No such file or directory
#include <stdio.h>
^
compilation terminate d.
/home/tuanle/RIOT/Makefile.base:64: recipe for target '/home/tuanle/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world/main.o' failed
make[1]: *** [/home/tuanle/RIOT/examples/hello-world/bin/stm32f4discovery/hello-world/main.o] Error 1
/home/tuanle/RIOT/examples/hello-world/../../Makefile.include:277: recipe for target 'all' failed
make: *** [all] Error 2

Can you help me for this issue , thank you very much

aabadie, 2017-05-05:
Hi,

This command works for me (Ubuntu + arm-none-eabi-gcc 4.9.3). You probably have missing development packages for correctly build RIOT on ARM platform.
You should follow [the doc](https://github.com/RIOT-OS/RIOT/wiki/Family:-ARM) for setting up your build environment ?

If you are running Ubuntu/Debian, you need to run the following first (including useful debugging tools):
```shell
sudo apt install build-essential g++-multilib openocd gcc-arm-none-eabi gdb-arm-none-abi binutils-arm-linux-gnueabi
```

closed 2017-10-25 by miri64

miri64, 2017-10-25:
(closed because not reproducable with proper set-up)