RIOT-OS/RIOT #7021: Nucleo f091 hello-world issue

nandra, 2017-05-08:
Hi, I'm trying to run hello-world on nucleo-f091 board. Basically I want to see some output on UART (should be 1152008N1) and I also enable following define:

```
diff --git a/boards/nucleo-f091/board.c b/boards/nucleo-f091/board.c
index b1244ac..814aa6e 100644
--- a/boards/nucleo-f091/board.c
+++ b/boards/nucleo-f091/board.c
@@ -20,6 +20,7 @@
 
 #include "board.h"
 #include "periph/gpio.h"
+#define AUTO_INIT_LED0
```

Problems it that I cannot see anything on console and also green LED is not enabled after make BOARD=nucleo-f091 flash

flashing output:
```
make BOARD=nucleo-f091 flash
Building application "hello-world" for "nucleo-f091" with MCU "stm32f0".

"make" -C /home/marek/projects/RIOT/boards/nucleo-f091
"make" -C /home/marek/projects/RIOT/core
"make" -C /home/marek/projects/RIOT/cpu/stm32f0
"make" -C /home/marek/projects/RIOT/cpu/cortexm_common
"make" -C /home/marek/projects/RIOT/cpu/cortexm_common/periph
"make" -C /home/marek/projects/RIOT/cpu/stm32_common
"make" -C /home/marek/projects/RIOT/cpu/stm32_common/periph
"make" -C /home/marek/projects/RIOT/cpu/stm32f0/periph
"make" -C /home/marek/projects/RIOT/drivers
"make" -C /home/marek/projects/RIOT/drivers/periph_common
"make" -C /home/marek/projects/RIOT/sys
"make" -C /home/marek/projects/RIOT/sys/auto_init
"make" -C /home/marek/projects/RIOT/sys/isrpipe
"make" -C /home/marek/projects/RIOT/sys/newlib
"make" -C /home/marek/projects/RIOT/sys/tsrb
"make" -C /home/marek/projects/RIOT/sys/uart_stdio
   text	   data	    bss	    dec	    hex	filename
   8540	    136	   2752	  11428	   2ca4	/home/marek/projects/RIOT/examples/hello-world/bin/nucleo-f091/hello-world.elf
/home/marek/projects/RIOT/dist/tools/openocd/openocd.sh flash
### Flashing Target ###
Open On-Chip Debugger 0.9.0 (2015-09-02-10:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v22 API v2 SWIM v5 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.251154
Info : stm32f0x.cpu: hardware has 4 breakpoints, 2 watchpoints
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f0x.cpu       hla_target little stm32f0x.cpu       halted
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x080003c8 msp: 0x20000200
auto erase enabled
Info : device id = 0x10006442
Info : flash size = 256kbytes
target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000003a msp: 0x20000200
wrote 10240 bytes from file /home/marek/projects/RIOT/examples/hello-world/bin/nucleo-f091/hello-world.hex in 0.467935s (21.370 KiB/s)
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x080003c8 msp: 0x20000200
target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0x20000200
verified 8676 bytes in 0.121442s (69.767 KiB/s)
shutdown command invoked
Done flashing
```

Before debugging can anybody confirm it's working on latest master (I'm on: 2ed677690158488d79f62b4cab01119c587c9f49)

aabadie, 2017-05-09:
It does work for me with master.

I'm using nucleo-f091RC. I don't think AUTO_INIT_LED0 is required except if you want to put it on in the main function for debugging. In this case, adding `#define AUTO_INIT_LED0` in board.c is not enough, you need to include `board.h` in main.c and call `LED0_ON;` macro in main.

Regarding your issue with UART, I would suggest that you update st-link firmware using [ST tools](http://www.st.com/en/embedded-software/stsw-link007.html). It solved these kind of problems several times.

nandra, 2017-05-09:
Thanks a lot for hint. Updating STlink FW helps.

closed 2017-05-09 by nandra