RIOT-OS/RIOT #7031: Packages: downloading code archives over http -> CI build may fail

MichelRottleuthner, 2017-05-10:
When working on a RIOT package (namely #6072 )that downloads its external code as an archive file from the authors webserver, I've seen some errors in the CI log regarding failing `$(DOWNLOAD_TO_FILE)` / `wget` calls. 
I could only observe this behaviour on superfast Murdock2 which may overload the used archive server with its parallel downloading. 
Are there any plans to add some caching to the `$(DOWNLOAD_TO_FILE)` mechanism?
What's the current best practice to reduce the number of code archive downloads on CI builds when using the [Makefile.http](https://github.com/RIOT-OS/RIOT/blob/master/pkg/Makefile.http) approach. 
Should the http-way be avoided in the future anyway?

smlng, 2017-06-03:
you mean like #6643

MichelRottleuthner, 2017-07-05:
@smlng ...better late than never: Yes! (I wonder what the heck I searched for to overlook this ;)

kaspar030, 2017-07-05:
I guess this can be closed.

closed 2017-07-05 by kaspar030