RIOT-OS/RIOT #7033: tests: unittests buildtests skipping

miri64, 2017-05-10:
When trying to run `make buildtest` on `tests/unittests` nothing is done. A bisect revealed a2b4c42c28425554a1c8083c3365be33016a77a7 to introduce that bug. BTW: how was this missed during testing for the release?

kaspar030, 2017-05-10:
> When trying to run make buildtest on tests/unittests nothing is done.

True that. Looking through the logs, Murdock1 didn't build the unittests since that commit.

> how was this missed during testing for the release?

Honestly, I didn't run the compile tests, as I relied on Murdock2, which correctly built everything for the corresponding commit. I guess I loosely misinterpreted the spec entry as "everything compiles", not "make buildtest works".

miri64, 2017-07-01:
@kaspar030 are you sure the unittests are compiling on Murdock? Because as far as I can tell they are currently broken for msp430-based platforms.

miri64, 2017-07-01:
(ah they are currently blacklisted via the `cbor` tests).

kaspar030, 2017-09-15:
@gebart The commit in question was yours, and honestly I don't understand it. Maybe you could take a look?

gebart, 2017-09-15:
The intention of the commit in question was to add FEATURES_REQUIRED += cpp when the unittests are built containing at least one C++ test module. For the platforms which don't support C++ we need to filter out the C++ test modules when building the complete test suite, to avoid dependencies on missing C++ compilers etc.

gebart, 2017-09-15:
I am not sure what part of it causes the problem though, possibly the FEATURES_REQUIRED part?

smlng, 2017-09-21:
FYI: when I run `make buildtest` in `tests/unittests` using latest master  on macOS it does compile for all boards, though all are reported as `failed`. Nevertheless, the original issue (*does nothing*) seems to be resolved?

gebart, 2017-09-21:
What is the failure? (run with BUILDTEST_VERBOSE=1)

smlng, 2017-09-21:
`BUILDTEST_VERBOSE=1` still only one line output

smlng, 2017-09-21:
the error is with the usage of `strptime` in `cbor.c`, if I change the Makefile for cbor to set `_XOPEN_SOURCE` for all (not only macOS) it works for all but AVR based MCUs. However, that is a separate issue - maybe cross compiling on macOS related?

kYc0o, 2017-10-20:
Is this issue still exists?

miri64, 2017-10-20:
I think not. Don't have a machine to test, but if `make buildtest` chose boards + status it should be ;)

miri64, 2017-10-25:
Yes, issue is fixed.

closed 2017-10-25 by miri64