RIOT-OS/RIOT #7045: when using module saul_adc: fatal error adc_params.h

raticate, 2017-05-11:
when I try to use the saul_adc module I get the following error:
fatal error: adc_params.h: No such file or directory
 #include "adc_params.h"

dylad, 2017-05-12:
I tested and got the same error message. 
I think this feature has not been implemented yet.

gebart, 2017-05-13:
Not all boards have a configuration file yet. You should write your own adc_config.h and place it in the boards/<BOARD>/include directory. You can copy the adc_config.h in boards/mulle/include for an example of what it is expected to contain.

smlng, 2017-07-06:
so the actual task would be: provide SAUL ADC adaption for all boards with feature `periph_adc`, right?

haukepetersen, 2017-10-25:
> so the actual task would be: provide SAUL ADC adaption for all boards with feature periph_adc, right?

not necessarily... We don't need to map everything just because we can, these mapping should be made in a sensible way I would say. So I'd say we go as we need them, makes sense?

haukepetersen, 2017-10-25:
@raticate which board are you using?

kYc0o, 2018-01-16:
Can we close this issue?

closed 2018-01-17 by raticate