RIOT-OS/RIOT #7052: ROP attack on RIOT

alexiss03, 2017-05-12:
Hi! 

I have replicated a `Return-oriented Programming (ROP)` attack in RIOT OS by using `ROPGadget` and by adding a CFLAG `-static -fno-stack-protector` in the Makefile of the application I created. Is it left intentional that the stack protector flag can be overriden? 

Thanks,
Lexi

gebart, 2017-05-12:
It is intentional that all CFLAGS should be possible to override from the project Makefile,  -fstack-protector and -fno-stack-protector are no special exceptions. If you would like to elaborate some more on this potential attack, could you please send an email to security@riot-os.org with some more information?

alexiss03, 2017-05-19:
Hi!

I'll send an email for this.

Thanks,
Lexi

miri64, 2017-05-22:
From what we've read on security@riot-os.org this is a vulnerability in the application, not the operating system, so I'm closing this issue. Re-open if you disagree.

closed 2017-05-22 by miri64

kaspar030, 2017-05-22:
> this is a vulnerability in the application, not the operating system

Well, Lexi just asked if it is intentional that the stack protector can be disabled at compile time, which I think @gebart answered perfectly.