RIOT-OS/RIOT #7058: tools: pyterm: EOF not handled correctly

kaspar030, 2017-05-14:
pyterm passes the EOF char (e.g., ctrl-d) as string "EOF" to the connected serial.
That causes problems with pexpect, which sends EOF e.g., when closing the spawned ```make term```.
I don't know exactly when it does that, but pyterm's behaviour breaks running pexpect scripts in a row.

```
/home/kaspar/src/riot.timers/dist/tools/pyterm/pyterm -p "/dev/ttyACM3" -b "115200"
Twisted not available, please install it if you want to use pyterm's JSON capabilities
2017-05-14 13:53:35,199 - INFO # Connect to serial port /dev/ttyACM3
Welcome to pyterm!
Type '/exit' to exit.
2017-05-14 13:53:36,965 - INFO # main(): This is RIOT! (Version: 2017.04-devel-1000-g5226a-booze)
2017-05-14 13:53:36,969 - INFO # RIOT network stack example application
2017-05-14 13:53:36,971 - INFO # All up, running the shell now
> 2017-05-14 13:53:55,532 - INFO #  EOF
2017-05-14 13:53:55,535 - INFO # shell: command not found: EOF
```

(I pressed CTRL-D at some point).

IMO pyterm should either ignore EOF, or send the EOF character, or exit, but definitely not send "EOF" as string.

OlegHahm, 2017-05-15:
If I remember correctly, the twisted package introduced this behavior.

kaspar030, 2017-10-12:
> If I remember correctly, the twisted package introduced this behavior.

Do you think it can be fixed?

aabadie, 2018-04-17:
I'm having a look, not sure if I can fix it

aabadie, 2018-04-17:
Opened #8970 to tackle this issue

PR #8970 (2018-09-04): dist/tools/pyterm: handle ctrl+d nicely

### Contribution description

This PR fixes the EOF being sent to the RIOT shell when one hits Ctrl+D in Pyterm.

The idea is to catch the 'EOF' string before it is sent to the shell and prompt the user if he wants to quit or not (there's the same behaviour in IPython).

If the user answers 'y' or 'Y', pyterm exits smoothly, if he answers 'n' or 'N', nothing happens, if he just hits 'Enter', this is considered as 'y' and pyterm exits and if he hits Ctrl+D a second time pyterm exits.

### Issues/PRs references

Fixes #7058
closed 2018-09-04 by aabadie