RIOT-OS/RIOT #7070: Makefiles: How to add subdirectories in an application directory?

kbumsik, 2017-05-17:
Hi, I am currently working on WINC1500 WiFi driver for MKR1000 and various boards (related: #6666)
I am trying to keep the code structure of the original WiFi driver in an application directory. So, my application directory looks like this:

```
app
  |-- main.c
  |-- boo1
  |     |-- boo1.c
  |     └-- boo1.h
  |-- boo2
  |     |-- boo2.c
  |     └-- boo2.h
  └-- boo3
        |-- boo3.c
        └-- boo3.h
```
and I added a following Makefile:
```
APPLICATION = test_winc1500
RIOTBASE ?= $(path-to-riot-root)

DIRS += boo1 boo2 boo3

BOARD ?= arduino-mkr1000

FEATURES_REQUIRED = periph_uart periph_spi periph_gpio 

USEMODULE += shell
USEMODULE += xtimer

include $(RIOTBASE)/Makefile.include
```
Lastly I added a simple `include $(RIOTBASE)/Makefile.include` Makefile in every `boo` directories.

When try compiling using `make all` it looks like each source is compiled but I got an error because they are not not linked with the main program at the last stage. At least I got compiled `boo*` objects in `bin/source` directory.

So I have a couple of question here:

1. Did I miss something? What script I should add to compile it correctly?
2. Does it look OK so far? Especially, do I need to add that simple Makefile in every subdirectory?

Thank you in advance,
Bumsik Kim.

gebart, 2017-05-17:
Try adding `USEMODULE+=boo1 boo2 boo3` in your Makefile. I think there is room for improvement here, but IIRC that is what is needed to add the files from subdirectories to the linking stage.

kbumsik, 2017-05-17:
Ohhh. It works! I now understand how MODULE is treated a little bit. Now I can communicate with the WiFi driver. Thanks a lot!

smlng, 2017-07-06:
closing, issue resolved.

closed 2017-07-06 by smlng

mrusme, 2019-08-09:
For everyone else who's struggling with this and found this issue via Google: This doesn't seem to work anymore the way it's described in here. However, there's a [working example inside the tests](https://github.com/RIOT-OS/RIOT/tree/master/tests/external_module_dirs). 👍