RIOT-OS/RIOT #7091: kw2xrf: RF Channel (19) of Death

cgundogan, 2017-05-23:
Tuning two `pba-d-01-kw2x` boards to RF channel 19 kills the communication between them. Pinging does not work between both boards (`gnrc_networking`). Can anyone else verify? @smlng could you give it a try?

gebart, 2017-05-23:
Did you try it at a different physical location? There could be some WiFi or other wireless activity in your area which interferes with it.

smlng, 2017-05-23:
I'm on it, I'll report back soonish ...

smlng, 2017-05-23:
cannot verify this:

```
> ifconfig
2017-05-23 10:14:14,718 - INFO #  ifconfig
2017-05-23 10:14:14,722 - INFO # Iface  7   HWaddr: 13:36  Channel: 19  NID: 0x23
2017-05-23 10:14:14,727 - INFO #            Long HWaddr: d3:c1:6d:42:ab:00:13:36 
2017-05-23 10:14:14,730 - INFO #            TX-Power: 0dBm  State: IDLE 
2017-05-23 10:14:14,735 - INFO #            ACK_REQ  AUTOCCA  MTU:1280  HL:64  6LO  RTR  IPHC  
2017-05-23 10:14:14,739 - INFO #            Source address length: 8
2017-05-23 10:14:14,742 - INFO #            Link type: wireless
2017-05-23 10:14:14,747 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2017-05-23 10:14:14,753 - INFO #            inet6 addr: fe80::d1c1:6d42:ab00:1336/64  scope: local
2017-05-23 10:14:14,758 - INFO #            inet6 addr: ff02::1:ff00:1336/128  scope: local [multicast]
2017-05-23 10:14:14,764 - INFO #            inet6 addr: ff02::1a/128  scope: local [multicast]
2017-05-23 10:14:14,765 - INFO #            
2017-05-23 10:14:14,768 - INFO #            Statistics for Layer 2
2017-05-23 10:14:14,772 - INFO #             RX packets 55  bytes 2395
2017-05-23 10:14:14,776 - INFO #             TX packets 18 (Multicast: 12)  bytes 689
2017-05-23 10:14:14,779 - INFO #             TX succeeded 17 errors 0
2017-05-23 10:14:14,782 - INFO #            Statistics for IPv6
2017-05-23 10:14:14,785 - INFO #             RX packets 17  bytes 1016
2017-05-23 10:14:14,790 - INFO #             TX packets 18 (Multicast: 12)  bytes 1066
2017-05-23 10:14:14,793 - INFO #             TX succeeded 18 errors 0
2017-05-23 10:14:14,794 - INFO # 
> ping6 fe80::d1c1:6d70:ab59:1336
2017-05-23 10:14:37,891 - INFO #  ping6 fe80::d1c1:6d70:ab59:1336
2017-05-23 10:14:37,905 - INFO # 12 bytes from fe80::d1c1:6d70:ab59:1336: id=84 seq=1 hop limit=64 time = 6.382 ms
2017-05-23 10:14:38,919 - INFO # 12 bytes from fe80::d1c1:6d70:ab59:1336: id=84 seq=2 hop limit=64 time = 6.384 ms
2017-05-23 10:14:40,921 - INFO # ping timeout
2017-05-23 10:14:40,925 - INFO # --- fe80::d1c1:6d70:ab59:1336 ping statistics ---
2017-05-23 10:14:40,931 - INFO # 3 packets transmitted, 2 received, 34% packet loss, time 3.0629831 s
2017-05-23 10:14:40,934 - INFO # rtt min/avg/max = 6.382/4.255/6.384 ms
```

smlng, 2017-05-23:
btw. used `gnrc_networking` from latest RIOT master, and board `pba-d-01-kw2x` (the newer revision with AA batteries).

smlng, 2017-05-23:
I did some further tests regarding this channel 19 issue, all successful:

- using different PAN ID
- ping between 2 PhyNodes
- ping from both PhyNodes to RasPi (with OpenLabs)
- ping from RasPi to both Phynodes

Some packet loss, but nothing serious. I'll recommend closing this issue as invalid - though there might be other networking issues present.

cgundogan, 2017-05-23:
I will give it another try later today with @gebart's recommendation

gebart, 2017-06-02:
@cgundogan Did you?

smlng, 2017-06-02:
@cgundogan I still think this is not a general issue, but has rather something to do with your specific test setup, hardware, and software - so close this?

gebart, 2017-06-04:
@cgundogan try #7139 #7138 and RIOT-OS/applications#30 for checking the radio environment. 
It does require an implementation of NETOPT_LAST_ED_LEVEL, which I haven't written for the kw2xrf, but it should be very similar to the implementation for kw41zrf in https://github.com/gebart/RIOT/commit/df58e8149d242303c0fc704aa164ad6b46fbb3eb

cgundogan, 2017-06-04:
@gebart I will try, thanks. I didn't look further into this issue, but I need to focus on this the following week. Will report back tuesday/wednesday on this.

smlng, 2017-07-06:
any news on this? Or stated differently: can this be closed?

PeterKietzmann, 2017-11-28:
@x3ro would you give it a try at the Hack'n'ACK tonight?

x3ro, 2017-11-28:
I've tried all sorts of pinging using the current `master` and `gnrc_networking` example. Everything seems to be working as expected, couldn't find any channels of ☠️. I'm closing this after conferring with @PeterKietzmann, since apparently it cannot be reproduced.

closed 2017-11-28 by x3ro