RIOT-OS/RIOT #7100: regression: assert check fails for SAMD21 ADC

photonthunder, 2017-05-25:
The assert in _adc_configure:

`assert(res >= ADC_RES_8BIT && res <= ADC_RES_12BIT);`

doesn't work for samd21 because ADC_RES_12BIT is 0x00 and ADC_RES_8BIT is 0x30.

Assert must have been turned on by default in a recent push or something, because it was not caught during quick tests.  Is dropping the check the best solution?

photonthunder, 2017-05-26:
Just looked at the datasheet and the saml21 has a similar feature, where 12Bit is 0 and 8Bit is 3.

dylad, 2017-05-29:
@photonthunder I just tried the lasted commit on master with SAML21-XPRO and tests/periph_adc (10 bits & 12 bits resolution). Everything was fine.
Could you describe a bit more your problem ?

dylad, 2017-05-29:
@photonthunder I looked more closely. Indeed, there is something wrong with `assert(res >= ADC_RES_8BIT && res <= ADC_RES_12BIT)` 
The check is wrong but assert was not catch when I tried.

I think we should just patch the assert and explain why we're doing like that with a dedicated comment.

photonthunder, 2017-05-30:
It would be a little ugly but we could just do `assert(res == ADC_RES_8BIT || res == ADC_RES_10Bit || ADC_RES_12BIT)`  Then the register values don't come into play if later a cpu is added that does things differently?

travisgriggs, 2017-05-30:
I don't think it's ugly. I think it's actually more clear and explicit and more safe. If the domain involved 5 or more linear arranged values, then maybe the comparators would be an appropriately necessary algorithmic approach. But where it's only these three, and it's causing fatal errors as it is right now, I like your solution @photonthunder . PR?

photonthunder, 2017-05-31:
[PR #7118 ](https://github.com/RIOT-OS/RIOT/pull/7118)

closed 2017-06-08 by photonthunder