RIOT-OS/RIOT #7109: help me to install RIOT in Arduino UNO

sravan7, 2017-05-30:


miri64, 2017-05-30:
Have you seen http://doc.riot-os.org/#the-quickest-start?

miri64, 2017-05-30:
(just set the `BOARD` environment variable to `arduino-uno`)

sravan7, 2017-05-30:
@miri64 yes I read quick start guide, but i can't able to do it. 
What I did ? 
1. I cloned Riot to my ubuntu using Terminal
2. After that I tried lot but i can't able to install Riot to Arduino Uno ,

Can you please help me step by step through screen shots or theough Video. 
Anyway Thank you.

sravan7, 2017-05-30:
And I tried   $ make BOARD = Arduino - uno 
result : No Bash , some thing is coming

PeterKietzmann, 2017-05-30:
you tried "BOARD=arduino-uno make"?

PeterKietzmann, 2017-05-30:
... with a small *a* and without white spaces... Furthermore, 'make' will only build the program, 'flash' will write the binary to your board and 'term' will open a serial connection to you can see the STDOUT in a terminal. You can simply do everything in one step copy-pasting the following line:
`BOARD=arduino-uno make all flash term`

kYc0o, 2017-05-30:
BTW @PeterKietzmann 's example command will only work on RIOT/examples/hello-world. Since the UNO board has only 2K of RAM, other examples may not fit into this amount of memory. What is worse, the compiler/linker won't throw any warning or error about memory overflowing.

miri64, 2017-05-30:
@kYc0o that might have been a little confusing. This also works with the applications in the Tutorial repository, but the memory constraint still holds.

sravan7, 2017-05-30:
No I tried all, didn't worked. Please help me from beginning to install RIOT into Arduino-Uno.

miri64, 2017-05-30:
Sorry, but unless you show us your problem, we can really help you. A start could be, if you might provide us with a dump of your terminal output (in most OSs you can just mark and copy it) from all the steps you have done so far.

PeterKietzmann, 2017-05-30:
Please provide a dump of your terminal output  (simply copy paste the text from you terminal where you typed the command BOARD=...)

PeterKietzmann, 2017-05-30:
whoops, didn't refresh the window...

sravan7, 2017-05-31:
sravan@sravan:~/RIOT$ cd example/default/
bash: cd: example/default/: No such file or directory
sravan@sravan:~/RIOT$ cd examples
sravan@sravan:~/RIOT/examples$ ls
arduino_hello-world  dtls-echo           gnrc_minimal     ipc_pingpong      posix_sockets
bindist              emcute              gnrc_networking  javascript        riot_and_cpp
ccn-lite-relay       gcoap               gnrc_tftp        microcoap_server  saul
default              gnrc_border_router  hello-world      nanocoap_server   timer_periodic_wakeup
sravan@sravan:~/RIOT/examples$ cd arduino_hello-world
sravan@sravan:~/RIOT/examples/arduino_hello-world$ ls
hello-world.sketch  Makefile  README.md
sravan@sravan:~/RIOT/examples/arduino_hello-world$ make all term
There are unsatisfied feature requirements: arduino

EXPECT ERRORS!

Building application "arduino_hello-world" for "native" with MCU "native".

In file included from /usr/include/limits.h:25:0,
                 from /usr/lib/gcc/x86_64-linux-gnu/5/include-fixed/limits.h:168,
                 from /usr/lib/gcc/x86_64-linux-gnu/5/include-fixed/syslimits.h:7,
                 from /usr/lib/gcc/x86_64-linux-gnu/5/include-fixed/limits.h:34,
                 from /home/sravan/RIOT/drivers/include/periph/gpio.h:56,
                 from /home/sravan/RIOT/sys/arduino/include/arduino.hpp:25,
                 from /home/sravan/RIOT/examples/arduino_hello-world/_sketches.cpp:1:
/usr/include/features.h:367:25: fatal error: sys/cdefs.h: No such file or directory
compilation terminated.
/home/sravan/RIOT/Makefile.base:87: recipe for target '/home/sravan/RIOT/examples/arduino_hello-world/bin/native/arduino_hello-world/_sketches.o' failed
make[1]: *** [/home/sravan/RIOT/examples/arduino_hello-world/bin/native/arduino_hello-world/_sketches.o] Error 1
/home/sravan/RIOT/examples/arduino_hello-world/../../Makefile.include:297: recipe for target 'all' failed
make: *** [all] Error 2
sravan@sravan:~/RIOT/examples/arduino_hello-world$ cd ..
sravan@sravan:~/RIOT/examples$ cd ..
sravan@sravan:~/RIOT$ BOARD=aduino-uno make all flash term
Welcome to RIOT - The friendly OS for IoT!

You executed 'make' from the base directory.
You should run 'make' in your application's directory instead.

Please see our Quick Start Guide at:
    https://github.com/RIOT-OS/RIOT/wiki/Quick-Start-Guide
Or ask questions on our mailing list:
    users@riot-os.org (http://lists.riot-os.org/mailman/listinfo/users)

Makefile:6: recipe for target 'all' failed
make: *** [all] Error 1
sravan@sravan:~/RIOT$ 
[image]

![screenshot

 from 2017-05-31 03-29-58](https://cloud.githubusercontent.com/assets/22323074/26616719/5a155884-45ee-11e7-8edb-6c186390b3c1.png)

miri64, 2017-05-31:
(took it upon me to fix your markdown)

miri64, 2017-05-31:
First of all, please make sure you installed all required dependencies:

**For Arduino:**
```bash
sudo apt-get install avr-libc gcc-avr avrdude
``` 

**For native** (i.e. run RIOT as a native Linux process):
```bash
sudo apt-get install multilib-gcc
``` 

You've typed `make all term`, not `BOARD=arduino-uno make all flash term` for `examples/arduino_hello-world`. Then you go to the root directory (which does not contain an application) and try to flash from there. Please go back to the `examples/arduino_hello-world` and type `BOARD=arduino-uno make all flash term`.

sravan7, 2017-06-01:
yeah.....! It's working. I wonder that's all installing RIOT in arduino  - uno is over ?.
 BOARD = arduino-uno make all flash term only one commnad. RIOT is for IOT where does COAP, LORAWAN , how to make use of them

gebart, 2017-06-01:
This discussion should be moved to the riot-users mailing list, it is not about a bug or issue in RIOT.
https://lists.riot-os.org/mailman/listinfo/users

miri64, 2017-06-01:
> yeah.....! It's working. I wonder that's all installing RIOT in arduino - uno is over ?.
BOARD = arduino-uno make all flash term only one commnad. RIOT is for IOT where does COAP, LORAWAN , how to make use of them

RIOT is the operating system. Network protocols you can add by adding modules. Please read the documentation on that. For LoRaWAN you need a special radio (see #6797) and I think there isn't enough space on an arduino uno for CoAP. If you want to try regardles, try one of the applications `examples/gcoap`, `examples/microcoap_server`, or `examples/nanocoap_server`. Each examplifies a different implementation of CoAP. Maybe one fits (not that you need to create your own application regardless if you still want to use the arduino libs).

> This discussion should be moved to the riot-users mailing list, it is not about a bug or issue in RIOT.
https://lists.riot-os.org/mailman/listinfo/users

@gebart is right. This kind of discussion usually belongs to the users mailing list. Would you be so kind to continue discussing this there, please?

kaspar030, 2017-06-01:
> I think there isn't enough space on an arduino uno for CoAP.

Challenge accepted! ;)

PeterKietzmann, 2017-06-01:
I see this issue as solved -> closing.

closed 2017-06-01 by PeterKietzmann