RIOT-OS/RIOT #7123: cpu: implement the stmlck interface for all STM32 CPUs

haukepetersen, 2017-06-01:
Lately, we introduced an interface for controlling the internal clocks for all STM32 CPUs (`stm32_common/include/stmclk.h`). This is however only implemented for the STM32F1 and STM32F7 so far and should be implemented for all members of that family. As result, problems like the one in #6504 are trivial to solve...

This issue is created to track and synchronize the porting effort.

Todo:
- [ ] stm32f0
- [ ] stm32f2
- [ ] stm32f3
- [ ] stm32f4
- [ ] stm32l0
- [ ] stm32l1
- [ ] stm32l4

haukepetersen, 2017-08-21:
As explicit reference: we should watch the outcome of #7477 to make sure there is no redundant work done when implementing this interface...

aabadie, 2018-04-13:
All stm32s cpus now support stmclk interface (#8735 did that for L0/L1), so we can close this one.

closed 2018-04-13 by aabadie