RIOT-OS/RIOT #7127: csma_sender is not being CI tested

gebart, 2017-06-02:
In trying to find out how to use the csma_sender module for software CSMA (see [mailing list thread](https://lists.riot-os.org/pipermail/users/2017-May/000818.html)), I noticed that it is not used as a module in any of the examples, nor is it a dependency of anything else in the repo. This means it will not be built by the CI tests and there is no standard way of testing it. A quick compilation test reveals that the module still builds correctly, but I have not figured out how I am supposed to use it together with my radio driver so I haven't tested running it. 

The optimal solution IMO would be to write a unit test for the csma_sender public functions, it should be possible to implement with a very basic netdev mock device.

miri64, 2017-06-02:
Could one of the assigned people provide a test application, please.

miri64, 2017-06-02:
The original plan seemed to have been that these would have been used in #4180, #4184, and #4213, but all of them are closed now. Is this module still needed at all?

gebart, 2017-06-02:
I would like to use it with the kw41zrf driver to provide csma since the hardware doesn't do it.

miri64, 2017-06-02:
@gebart Both functions provided in this module should be straight forward to be used instead of `netdev->driver->send()` in your application or network stack glue code, as far as I can remember. For good parameters for CSMA/CA you need to ask the MAC experts I assigned this issue to ;-).

gebart, 2017-06-02:
@miri64 Actually, I was thinking about adding it [here](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/link_layer/netdev/gnrc_netdev_ieee802154.c#L224)

miri64, 2017-06-02:
Fine by me, but only optional (with `#ifdef MODULE_CSMA_SENDER`) I hope. ;-)

miri64, 2017-06-02:
@gebart also please note https://github.com/RIOT-OS/RIOT/pull/6554#issuecomment-305756434

zhuoshuguo, 2017-06-02:
> Could one of the assigned people provide a test application, please.

I will see if I have time to do it. In case I have enough spare time, will try to write a test application.

zhuoshuguo, 2017-06-02:
> For good parameters for CSMA/CA you need to ask the MAC experts I assigned this issue to ;-).

Normally, in case your test scenario is no too intensive, a default CSMA parameter in ` csma_sender`, as far as I remember (BE-min of 3, and BE-max of 5), will be OK, for example test scenario like: one receiver and 2-3 senders, with data rate 1packet/1s, or even multi-hop network. :-)

zhuoshuguo, 2017-06-30:
Maybe this issue is settled now with the merge of [LWMAC](https://github.com/RIOT-OS/RIOT/pull/6554) which includes `csma_sender`. :-)

miri64, 2017-06-30:
Agreed.

closed 2017-06-30 by miri64