RIOT-OS/RIOT #7133: Usage and interpretation of I2C addresses

smlng, 2017-06-03:
I recently tested the LSM6DSL sensor (part of STM X-NUCLEO IKS01A1), which provides accelerometer and gyroscope. During configuration I stumbeled upon the following addressing issue:

The default I2C address of the RIOT driver is `0x6B` [1].  According to a label printed on IKS01A1 the default I2C address of LSM6DSL is `0xD6`, see also [2] page 38, sec 6.3.1. However, the driver worked right out of the box, no need to change the address config. 

I'm not an I2C expert (neither in general nor for RIOT 😉 ). However, fun fact is that

- `0x6B` = `0110` `1011` 
- `0xD6` =  `1101` `0110`

As you see, they are essential the same just reversed. Is this (kind of) standard, known, or to be expected?

[1]: https://github.com/RIOT-OS/RIOT/blob/master/drivers/lsm6dsl/include/lsm6dsl_params.h#L39

[2]: http://www.st.com/content/ccc/resource/technical/document/datasheet/ee/23/a0/dc/1d/68/45/52/DM00237456.pdf/files/DM00237456.pdf/jcr:content/translations/en.DM00237456.pdf

gebart, 2017-06-03:
I don't think it is reversed, but shifted one bit. The least significant bit of the i2c address is the read/write flag, so the actual address is in reality only 7 bits long. Does this make sense? 
See also this link https://www.totalphase.com/support/articles/200349176-7-bit-8-bit-and-10-bit-I2C-Slave-Addressing

smlng, 2017-06-03:
Thanks for the pointer, could be the shifted. 

However, still it seems weird, that the device label says default address is `0xD6` and the driver uses `0x6B` (which works). At least for I2C noobs like me, simply trying to use the labeled address will likely result in lots of frustration ☹️

smlng, 2017-06-03:
mhm, seems to be shifting, works for the alternative addresses, too

- driver: `0x6A` = `0110` `1010`
- device label: `0xD4` = `1101` `0100`

still as said above, not very intuitive

aabadie, 2017-06-04:
>  LSM6DSL sensor (part of STM X-NUCLEO IKS01A1)

In fact, LSM6DSL is part of [STM X-NUCLEO IKS01A2](http://www.st.com/en/ecosystems/x-nucleo-iks01a2.html) and it is LSM6DS**0** which is part of [STM X-NUCLEO IKS01A1](http://www.st.com/en/ecosystems/x-nucleo-iks01a1.html).

From the datasheets, the sensors are sensibly different (I tried when I reviewed #6835)

smlng, 2017-06-04:
@aabadie, I know there are different version of this sensor - but here it just serves as an example for the counter-intuitive (not matching) addressing between RIOT driver and device datasheet (or labeling).

gebart, 2017-06-04:
@smlng I understand how you think it is unintuitive, but there's actually a variation between how different device vendors write their data sheets. Some write the 8 bit address including the read but, and other vendors write the 7 bit address and a short note that the eighth bit is a read/write flag.

smlng, 2017-06-06:
@gebart thanks a lot for your input and pointers, I think I got it ... And you were right with the *shifting* explanation. 

To that matter I'll review the RIOT I2C documentation, maybe be I can provide some enhancements to clarify.

smlng, 2017-06-06:
I'll close the issue after I checked the docu, in general its resolved for me.

smlng, 2017-06-08:
closing, fixed with #7142 merged

closed 2017-06-08 by smlng