RIOT-OS/RIOT #7140: How to add additional files to my project ?

Marc-Aurele, 2017-06-04:
Hello,

I want to add additional c files to my application project but i don't really know how to do that. In my main application folder, where i have already the following files : 
_main.c
Makefile
README.md_

I want to add a "src" folder and an "inc" folder to the compilation. So I have added a makefile to the src folder with the following contain:

_MODULE = main_app_src
include $(RIOTBASE)/Makefile.base_

And in the Makefile that is with my main.c file, i have added the following lines:

_USEMODULE += main_app_src
INCLUDES += -I$(CURDIR)/inc
DIRS += $(CURDIR)/src_

Do you agree with that or do i need to do in a different way ?

Thank you very much.

Aurélien

gebart, 2017-06-04:
Your solution looks correct. Did you test it?
In the future, these kinds of questions are better to ask on the riot-users mailing list. This issue tracker is meant for issues in RIOT. You will likely get better and more complete answers on the mailing list. 
https://lists.riot-os.org/mailman/listinfo/users

Marc-Aurele, 2017-06-04:
Hi,
Thank you for the reply.
Yes everything is working fine. Sry for having posted my question here, next time i will use the mailing list ;)